iNavFlight/inav #4028: Fixed Wing Output Mapping in Inav 2.0 ?

XenXwing, 2018-11-21:
Hi
Newbie here and have question about Mix > Output Mapping on Inav 2.0
My very first Fixed wing set up with F3 Colibri V2, soldered S1, 3, 4. 
 - S1 Motor
 - S2 Empty
 - S3 Servo 3
 - S4 Servo 4

But my result was that only one servo (left) is working..
When I check Mix Tab, mapping seems to be wrong and I do not know what should I do?

Please check my attachment, showing Inav Mix tab.
thanks
[image]

shellixyz, 2018-11-21:
This target only supports one motor output so the output mapping table is right you need to connect the servos to ouputs 2 and 3.

XenXwing, 2018-11-22:
Thanks

> This target only supports one motor output so the output mapping table is right you need to connect the servos to ouputs 2 and 3.

DzikuVx, 2018-11-22:
since there is nothing else to do, I'm closing. Thanks

closed 2018-11-22 by DzikuVx

666satyr666, 2019-03-14:
Please tell me how you doing it @XenXwing.
I got a unique case so my FC only remaining 3 servo output, just like what I need for flying wing. But I can't remapping those pin. I need to be just like what you set in here.
I need pin #1 for motor and pin #2 & #3 fro servos.