iNavFlight/inav #4030: No resource for SDA SCL solder pads on DYS F4 Pro V2

Schakus, 2018-11-22:
Hey,
i want to use the I2C for a barometer but there is no resource. Resource mapping is not implemented in inav so i can't do it by myself. I think this describes it: [betaflight-issue](https://github.com/betaflight/betaflight/issues/5127)
Could somebody help me please?
Thanks!

`# resource
IO:
----------------------
A00: FREE 
A01: MOTOR5 OUT
A02: MOTOR4 OUT
A03: MOTOR3 OUT
A04: MPU CS
A05: SPI1 SCK
A06: SPI1 MISO
A07: SPI1 MOSI
A08: MOTOR6 OUT
A09: SERIAL1 UART TX
A10: SERIAL1 UART RX
A11: USB IN
A12: USB OUT
A13: FREE 
A14: FREE 
A15: OSD CS
B00: MOTOR1 OUT
B01: MOTOR2 OUT
B02: FREE 
B03: FLASH CS
B04: BEEPER OUT
B05: LED1 OUT
B06: FREE 
B07: FREE 
B08: FREE 
B09: FREE 
B10: SERIAL3 UART TX
B11: SERIAL3 UART RX
B12: FREE 
B13: FREE 
B14: FREE 
B15: FREE 
C00: INVERTER OUT
C01: ADC CH1
C02: ADC CH2
C03: FREE 
C04: MPU EXTI
C05: FREE 
C06: SERIAL6 UART TX/RX
C07: FREE 
C08: FREE 
C09: FREE 
C10: SPI3 SCK
C11: SPI3 MISO
C12: SPI3 MOSI
C13: FREE 
C14: FREE 
C15: FREE 
D00: FREE 
D01: FREE 
D02: FREE 
D03: FREE 
D04: FREE 
D05: FREE 
D06: FREE 
D07: FREE 
D08: FREE 
D09: FREE 
D10: FREE 
D11: FREE 
D12: FREE 
D13: FREE 
D14: FREE 
D15: FREE `

giacomo892, 2018-11-22:
What about connecting the baro to the i2c port? It should work.

Schakus, 2018-11-22:
@giacomo892 it's physical connected but missing in the resource's. That's why it's not working.

giacomo892, 2018-11-22:
Should work on the UART3 pads.

Schakus, 2018-11-22:
But I need the UART and want to use the i2c pads.

shellixyz, 2018-11-23:
You can't use both UART3 and I²C. They are shared.

Schakus, 2018-11-23:
Are you sure? because the link from the betaflight ticket says i2c is on 
```
resource i2c_scl 2 b8
resource i2c_sda 2 b9
```

and in inav these are free:
```
B08: FREE 
B09: FREE 
```

Schakus, 2018-11-23:
And that would be very annoying because then i can't use my gps, smartaudio, f-port and barometer at the same time...

digitalentity, 2018-11-23:
Question is where are those PB8/9 connected on DYS F4? What would be the
point of I2C of you can't use it?

Schakus, 2018-11-23:
The I2C pins are **not** parallel to the UART3 (i measured it). So it should possible to use them.

If i could not use i2c i can't fly automatically because the baro is missing.

giacomo892, 2018-11-23:
There should be i2c pads on the top of the board but currently we share the target with OMNIBUSF4 and so we share I2C with UART3. 
A new fix/ new target would be required to map I2C PB8/9

digitalentity, 2018-11-23:
Indeed the V2 seems to have pads of I2C. We should create a separate target
as @giacomo892 suggested.

Schakus, 2018-11-23:
Thanks! Exactly what I mean. Do you have capacity to do this in the next time?
The drone waits for the maiden :-)

giacomo892, 2018-11-23:
@Schakus does this boots and I2C works on the I2C pads?
[inav_2.1.0_DYSF4PRO.hex.zip](https://github.com/iNavFlight/inav/files/2610481/inav_2.1.0_DYSF4PRO.hex.zip)

Schakus, 2018-11-23:
@giacomo892 The FC is working but the I2C not. I also tried it with betaflight but its not working with the following command:
```
resource i2c_scl 2 b8
resource i2c_sda 2 b9
```
How can i check the pin mapping?

Schakus, 2018-11-23:
I double checked it. Pin B8 and B9 is the right pins. Do you have any ideas?
[pinmapping](http://waijung.aimagin.com/hmfile_hash_0f62a7f9.jpg)

Schakus, 2018-11-23:
I measured with an oscilloscope but there is no traffic on both lines. Both are on 3.3V

shellixyz, 2018-11-23:
Please try this build: [inav_2.0.1_DYSF4PRO.hex.zip](https://github.com/iNavFlight/inav/files/2611886/inav_2.0.1_DYSF4PRO.hex.zip)

Schakus, 2018-11-24:
@shellixyz it works! Thank you! Will you add it to the official release?

shellixyz, 2018-11-24:
Great! Sure we will

shellixyz, 2018-11-25:
A new target DYSF4PROV2 will be available in 2.1. Fix ready to merge, closing.

closed 2018-11-25 by shellixyz