iNavFlight/inav #4032: GPS connection problem in INAV2.0.1

DusKing1, 2018-11-23:
## Current Behavior

GPS info sometimes red and sometimes blue in configurator. Can not arm.
## Steps to Reproduce

1.Flash INAV2.0.1
2.Connect uibox7 to FC
3.Power the system
4.Connect FC to configurator

## Additional context

- FC Board name and vendor: MatekF722
- INAV version string: 2.0.1
- Everything works good when flash back iNav2.0.0

giacomo892, 2018-11-23:
Be sure the TX line is firmly connected to the GPS unit and also you might also need to wait a bit in order to the GPS modules to the configured. 
In INAV 2.1 timeouts will be relaxed to allow GPS to have to more time to be configured properly.

digitalentity, 2018-11-27:
UART doesn't matter, if UBX GPS is not recognized this means that bidirectional comms with it is not possible.

Mochaboy, 2018-11-27:
I agree it shouldn't, but it did in my case.  I brute forced the connection scenarios, enabled gps on uart 1, 3, and 6, tried every combo of tx->tx, rx-> rx, tx->rx, rx-> tx, and the only connection that worked on an HGLRC F4 Board was Uart 6.  

I've seen this exact same behavior on other omnibus F4 boards and I couldn't make sense out of why certain uarts worked and others did so now it's just a general part of my troubleshooting.  I even went so far as to look up the PB pins in the code for a given target board and soldered the tx/rx wires directly to the STM32 pins to no effect.  

This is all after verifying via u-center that the gps pucks work.  I've even gone so far as to reset gps pucks in ucenter to the default configurations (9600 baud, 1hz, pedestrian mode, no galileo etc...)...

I thought it was a curious release note because the way it reads, it sounds like in certain cases, pin mappings were changed which would absolutely force a 2.00 to 2.01 installer to have to switch rx/tx connections.

romko1, 2018-12-05:
After updating FW to INAV 2.0.1, it takes a long time to get a GPS 3D fix (cold start: 5-8min).
 I have a FC MATEKF405 -> UART5 (TX/RX) for GPS, GPS module: Beitian BN-880. When I had FW 1.9.1 or 2.0.0 so the cold start was shorter (2-3min.)
 I have a BAUT-rate of 57600. When connecting GPS to u-center, I have a 3D fix within max. 2 minutes indoor.
 I am sure I have wired GPS module correctly.
My setting:
set gps_provider = UBLOX
set gps_sbas_mode = AUTO
set gps_dyn_model = AIR_1G
set gps_auto_config = ON
set gps_auto_baud = ON
set gps_ublox_use_galileo = OFF
set gps_min_sats = 6

shellixyz, 2018-12-05:
@romko1 I doubt it has anything to do with iNav. There is probably conducted noise from the power supply wires or radiated noise from other peripherals in your setup interfering with the reception of the GPS module.

romko1, 2018-12-05:
@shellixyz: It also occurred to me, but I did not modify the hardware (position of the GPS module or cables) when changing FW Inav 1.9.1 -> 2.0.0 -> 2.0.1. It seems to me that the new version of Inav 2.0.1 is more sensitive to receiving data from satellites and related 3D fix. Which is strange..

Mochaboy, 2018-12-05:
@romko1 I'm running a BN-220 on 2.01...went to the field yesterday, parked the car, plugged the bird, set it down for a cold start gps lock... By the time I got my goggles on and radio turned on (less than 2 minutes later), I already had 11 satellites locked.  I'll have to see if I can dig up a BN-880 to replicate your issue but it doesn't seem like it should matter (on the surface).

wx4cb, 2018-12-05:
try turning off autobaud and autoconfig. see if that's any better. providing you have already setup the baud rate etc correctly in the ucenter software.

romko1, 2018-12-05:
Thank you for advice. I tried to set the set gps_auto_config = OFF and set gps_auto_baud = OFF and GPS 3D fix time around 4:40 when setting auto_baud and auto_config ON time approximately the same.
I always put the copter on the same place on the ground and I'm waiting. 

For your information: Even with FW Inav 1.9.1 I had a very long time on a 3D fix, but I solved it by pulling the JST 2 pin connector from the GPS battery and after adding (when the battery is still charged) I attach the CR2032 battery. When I fly, I simply disconnect the CR2032 and so I managed to shorten the cold start within 2 minutes.

wx4cb, 2018-12-05:
that certainly sounds like a gps unit issue. have you tried a similar unit ?

romko1, 2018-12-06:
I do not have any other GPS. I'm thinking of ordering the mini Radiolink TS100, only when there are problems with the compass in INAV :(

wx4cb, 2018-12-06:
if you don't need a compass - which on a fixed wing you dont, then this is the one I use. I get full fix in the time it takes me to button the a/c up and put my goggles on: 

https://www.racedayquads.com/products/micro-ublox-m8n-gps-glonass

Mochaboy, 2018-12-06:
@romko1 I've done a lot of troubleshooting in this area with near field RF scanners.  You'd be surprised at the number of issues that crop up with out of band emissions radiating off of low quality poorly filtered video transmitters (enough to swamp a gps in the case of the earlier TS58xx vtx units).  I ran a test recently with the Runcam Split v1, v2 and Caddx Turtle v1, and v2 and within about 10 inches, of the runcam - that was enough to disrupt half of my satellite locks.  So video transmitters (5.8 and 1.3ghz I forget which channel but one's worse than the other for gps locks), high definition recorders, even some of the older flight cameras - all had issues that if left unaddressed, could swamp GPS Upper and Lower L Band frequencies and even 433 control frequencies.

romko1, 2018-12-06:
@wx4cb The GPS module does not look bad. Whether I need a compass? I use copter for RTH, Pos Hold and waypoints - missions. Is it necessary for these compass modes?

@Mochaboy I have a RunCam Split mini camera about 3cm from the GPS module, but the camera is in a carbon frame and the GPS module is still laid on an aluminum mat that goes beyond its size.

Mochaboy, 2018-12-06:
@romko1 Do you have an RF Explorer? - that's the only way to be sure your RF damping is working (other than brute forcing with u-center).  I have a video I posted that shows the effects of a Runcam Split on gps signals so you can see...the problem is also that it's not necessarily the hd camera itself that radiates noise - but the wiring as well...at least that's the near field scans showed.  Anyway easy way to troubleshoot would be to power the rig up without runcam and see how long a cold start takes.

romko1, 2018-12-06:
I do not have an RF explorer but it is not a problem to download a program from the Internet. But I do not have hardware. 
I made a small attempt this evening: Around the GPS antenna, I placed around 5-6 cm of aluminum foil and molded it into a parabolic shape. Cold start within 30 seconds and 10 satellites reported in Inav ... So I got interference from copter electronics. :( ...Of course, it was extreme, the copter could not fly.

shellixyz, 2018-12-06:
@romko1 I don't want to insult your intelligence but make sure the antenna of your GPS is pointing up (the ceramic square) we've seen quiet a few instances where people have installed their GPS module upside down and that of course doesn't help :smile:.

romko1, 2018-12-07:
@shellixyz That's fine, of course, the ceramic antenna is pointing upwards. With this aluminum foil I just wanted to point out that I prevented the interference of the GPS module from the electronics copter. When I connected the GPS module myself via the FTDI converter to the u-center, I had a 3D fix inside the window within 2 minutes. After connecting the GPS module to the FC copter, I outdoors in the open space to get 3D fix after 5-8 minutes.

tiriad, 2018-12-18:
> After updating FW to INAV 2.0.1, it takes a long time to get a GPS 3D fix (cold start: 5-8min).
> I have a FC MATEKF405 -> UART5 (TX/RX) for GPS, GPS module: Beitian BN-880. When I had FW 1.9.1 or [...]

Disable the sbas, can get faster the position

digitalentity, 2018-12-19:
Behavior differences:
2.0.0:

 - GPS settings are not enforced, GPS starts with pretty much random settings
 - Unsafe, but usually gets fast fix

2.0.1+:

 - GPS settings are enforced
 - Default settings require good satellite coverage and good GPS antenna
 - Slower fix, but more predictable GPS behavior

Settings to play with (at your own risk):

```
set gps_dyn_model = ...
set gps_sbas_mode = ...
```

closed 2019-03-03 by stale[bot]