iNavFlight/inav #4040: INAV Mavlink Datastream stops with crossfire on arming

wx4cb, 2018-11-27:
## Current Behavior
FC: Matek F405 Wing FC
INAV Version: 2.0.0 (and 2.0.1)
RC Link: TBS Crossfire full module on taranis (latest version) to crossfire micro.
Link Details: CRSF on UART 5, MAVLINK and MSP on UART3 @ 57600

Current behaviour is, I get full data coming down over the bluetooth link to the crossfire module (set to bridge on channels 3/4) As soon as I arm, I get no data - IE the data stream stops. INAV Mission planner on android just repeats (by voice) the last data interspersed with "no data or no connection". QGround Control does the same. Mobile flight on IOS does the same thing.

See attached video link. This has been verified on 2 matek F405 boards in 2 different models, 2 different IOS devices and 3 different android devices. 

## Steps to Reproduce

1. Setup crossfire bluetooth link
2. Connect with android or ios device
3. Get data
4. Arm
5. Lose Data

## Expected behavior
No data dropouts

## Suggested solution(s)
Not sure. Don't understand why

## Additional context
https://pastebin.com/MpR7Gu3N

INAV version string: INAV/MATEKF405SE 2.0.0 Aug 20 2018 / 19:22:34 (dbdd1656a)

wx4cb, 2018-11-27:
My  only question... do I need both mavlink AND msp? or just one or the other? btw, here's a video explaining the issue: https://www.youtube.com/watch?v=l4of7IcIwn4

wx4cb, 2018-11-27:
OK. Just out of curiosity, I disabled MSP on that port and it worked. except i wasn't getting GPS data. 

When I disabled Mavlink on that port and enabled MSP, it worked fine. So I think there's an issue with the mavlink data stream somewhere along the line as it's not sending out the gps coordinates or anything else other than pitch/roll etc

wx4cb, 2018-11-27:
Not saying its an inav issue at all specifically, im just trying to rule out a pebkac (user) error.

@stronnag Do you have time to post your ports tab for me?

stronnag, 2018-11-27:
I'm just trying to eliminate complexity (and closed source tools that preclude analysis) .... my 3DR link is set up with serial speed 115200 and air speed 64k, so close to 57600 over the air. It is a pure transparent serial (no MAVLINK framing) as I normally prefer LTM.

```
# serial
serial 20 1 115200 38400 115200 115200
serial 0 257 115200 38400 115200 115200
serial 1 64 115200 115200 0 115200
serial 2 0 115200 115200 0 115200
serial 3 2 115200 115200 0 115200
serial 4 0 115200 115200 0 115200

# version
# INAV/MATEKF405 2.1.0 Nov  8 2018 / 19:01:54 (f740c47c1)
# GCC-8.2.0
```
[image]

wx4cb, 2018-11-27:
Hmm interesting. So set them to 115.2 as opposed to 57600?

Also, use map and ltm? The crossfire is acting as a serial bridge

stronnag, 2018-11-28:
The speed is not really relevant as long as it's fast enough to sustain MAVLINK (needs at least 19200), Otherwise, it's up to you and the GCS you are using.

closed 2018-11-28 by wx4cb