iNavFlight/inav #4043: Buzzer Sequence selection

kardon18, 2018-11-29:
## Current Behavior
Buzzer is enabled by default on platforms that have buzzer connections.
All tones are played when ever any of the following are noted -
* Low and critical battery alarms (when battery monitoring enabled)
* Arm/disarm tones (and warning beeps while armed)
* Notification of calibration complete status
* TX-AUX operated beeping - useful for locating your aircraft after a crash
* Failsafe status
* Flight mode change
* Rate profile change (via TX-AUX switch)

Many of these tones can be annoying, when consistently sounding. 
While others like  - **TX-AUX operated beeping**. Are useful as a lost model alarm.

_Current behavior of these tones can not be selected.
They are all ON, by default._

## Desired Behavior
Allow user selection. 
 

## Suggested Solution
Add CLI commands to allow the individual tone sequences to be Enabled/Disabled according to user requirements. 

## Who does this impact? Who is this for?
It will impact anyone who wants to use a high dB buzzer for a lost model alarm. 
But value their hearing. And doesn't want to be deafened by continuous warnings of events that they are already aware of. Or they do not care about that specific warning. 

## Additional context
Many lost model alarm buzzers can reach 90dB. 
[Matek lost model buzzer.pdf](https://github.com/iNavFlight/inav/files/2628913/Matek.lost.model.buzzer.pdf)

shellixyz, 2018-11-29:
You can already enable/disable beeper events with the `beeper` command. See this example:
[image]

kardon18, 2018-11-29:
Thanks @shellixyz 
I look through the wiki everywhere and only found reference to "buzzer".
I didn't think to look for "beeper" in the CLI dump.

kardon18, 2018-11-29:
Thanks @shellixyz 
I looked through the Wiki everywhere and only found reference to "Buzzer".
I didn't think to look for "Beeper" and in the CLI dump.

patemil, 2018-12-11:
Hi and thanks for all the very good work!
I'm running Matek 405 CTR on a quad with fw from Aug 20 2018 release and neither -ON_USB nor -SYSTEM_INIT seems to work as expected. Always beeps at start and if USB is connected.

kardon18, 2018-12-19:
@patemil 
ON_USB nor -SYSTEM_INIT seems to work as expected. Always beeps at start and if USB is connected.

With `beeper -ON_USB`  To disable beeping when the FC is connected to USB. 
It still makes a power up beep, followed by a rapid succession of beeps.

The main idea of ON_USB mode. Is to prevent other warnings from sounding with a PC connection.
Which would drive us mad.

patemil, 2018-12-19:
@kardon18
Thanks for your answer, this is how I suspected it to work but with -ON_USB I still get the ones that are turned on such as battery warning. I have disabled most of the other beeps and that works.