iNavFlight/inav #4047: SBUS/PWM receiver doesn't work with AnyFC F7

golddragon007, 2018-12-01:
## Current Behavior
If I try to use PWM every signal on the receiver tab is 0,
If I try to use SBUS every signal on the receiver tab is 1500.
And none of these two cases I can do anything with it (I can't increase or decrease the values).

## Steps to Reproduce

1. Flash or reset (?).
2. Connect your receiver for SBUS or PWM usage.
3. Set the FC for SBUS or PWM usage.
4. Go to receiver tab and none of the valuse will be moved.

## Expected behavior
If I change a setting on the transmitter it should affect that page, if it's transmitted.

## Suggested solution(s)
Look what's changed between the 2.0.1 and 1.9.1. In 1.9.1 it worked both (PWM and SBUS).

## Additional context
https://pastebin.com/aUJGAG4s

---

- FC Board name and vendor: AnyFc F7
- INAV version string: INAV/ANYFCF7_EXTERNAL_BARO 2.0.1 Oct 31 2018 / 18:48:13 (a04e68aaa)
- Receiver R7008SB

nik012003, 2019-01-25:
Hi, I've the same problem on a Matek F722 board. Have you found a solution ?

shellixyz, 2019-04-04:
I tested S.Bus and PWM inputs on the AnyFC F7 with 2.1.0:
- S.Bus seems to work fine on UART3
- PWM inputs don't work

shellixyz, 2019-04-04:
PWM inputs won't be fixed unless there is some demand for it. Nobody uses PWM inputs anymore.

closed 2019-04-04 by shellixyz