iNavFlight/inav #4053: I have a FrSky X10S transmitter with a R9M module and a L9 receiver. In INAV I can't get any stick movement my Matek F405 wing controller seems to be ok. What do I wrong?

Jopie80, 2018-12-03:
For immediate help, just ask your question on one of the following platforms:

* [Telegram channel](https://t.me/INAVFlight)
* [Facebook group](https://www.facebook.com/groups/INAVOfficial)
* [RC Groups thread](https://www.rcgroups.com/forums/showthread.php?2495732-Cleanflight-iNav-%28navigation-rewrite%29-project)
* [IRC channel](irc://irc.freenode.net/#inavflight)
* [Slack channel](https://publicslack.com/slacks/inavflight/invites/new)

You can also read public documentations or watch video tutorials:

* [Official documentation](https://github.com/iNavFlight/inav/tree/master/docs)
* [Official Wiki](https://github.com/iNavFlight/inav/wiki)
* [Video series by Painless360](https://www.youtube.com/playlist?list=PLYsWjANuAm4qdXEGFSeUhOZ10-H8YTSnH)
* [Video series by Paweł Spychalski](https://www.youtube.com/playlist?list=PLOUQ8o2_nCLloACrA6f1_daCjhqY2x0fB)

teckel12, 2018-12-04:
My guess? Confusion between title and description?

Jopie80, 2018-12-04:
I have a Matek f405 wing flight controller and want to configure it with INAV in combination with a FrSky X10S transmitter with a R9M module (long range) and a R9 receiver. I configured it in INAV with the instructions on YouTube from Painless360 ( https://www.google.nl/search?rlz=1C1ZKTG_nlNL774NL774&ei=aG0GXOPCHtDSa_bzmrAB&q=painless360+inav&oq=painless3&gs_l=psy-ab.1.2.0j0i203l7j0j0i30.129878.134413..150182...1.0..0.136.648.9j1......0....1..gws-wiz.......0i71j35i39j0i131j0i67j0i10.ysobQmR6TWs ) . It seems that I made somewhere a mistake but I can't find it. I connected my receiver with S-Bus, made the changes in the Tab Ports. As I look in the Tab Receiver the slides won't move when I use my transmitter.