iNavFlight/inav #4055: Matek F722-STD - Can't Update Fonts - Board Reboots Every Time

CradyLab, 2018-12-03:
## Current Behavior

FC reboots on attempts to upload font files from within iNav - tested with Betaflight and Butterflight and they both work correctly

## Steps to Reproduce

1. Enable OSD, save & reboot
2. Go to OSD settings, setup desired display
3. Select desired font from font menu tabs
4. Click to upload selected font
5. Progress bar shows uploading and upon reaching 100% forces the board to reboot never giving the opportunity to click 'save'

## Expected behavior

Upon uploading the font file the board should not reboot and should have the chance to save your settings

## Suggested solution(s)

## Additional context

---

- FC Board name and vendor:
MATEK F722 STD
http://www.mateksys.com/?portfolio=f722-std

- INAV version string:

INAV/MATEKF722 2.0.1 Oct 31 2018 / 19:27:11

giacomo892, 2018-12-03:
You don't need to save. It's already done.

CradyLab, 2018-12-03:
> You don't need to save. It's already done.

When I go back into iNav the displayed fonts haven't changed - still look like the defaults

CradyLab, 2018-12-03:
I'm also see getting those dashed lines in the display with iNav that I don't get with Betaflight or Butterflight

shellixyz, 2018-12-03:
> When I go back into iNav the displayed fonts haven't changed - still look like the defaults

That's normal but the fonts on the OSD output itself should have changed

> I'm also still getting those dashed lines in the display with iNav that I don't get with Betaflight or Butterflight

What version of the configurator are you using ?

CradyLab, 2018-12-03:
Configurator v2.0.0

I can't verify the fonts on the FC have actually changed -- waiting on a battery to finish charging

CradyLab, 2018-12-03:
> That's normal but the fonts on the OSD output itself should have changed
>
>
> What version of the configurator are you using ?

Thanks, man.  It all seems to be working now =)

closed 2018-12-03 by CradyLab