iNavFlight/inav #4058: Feature request: allow slower BB logging

shellixyz, 2018-12-04:
## Current Behavior

Max BB rate denom is 255 meaning minimum rate of 16Hz for 4KHz loop rate

## Desired Behavior

Slower BB logging rates for high rate loop settings
Asked by Olivier C

## Suggested Solution

Increase `blackbox_rate_denom` setting's max value

shellixyz, 2018-12-20:
I haven't done anything about this yet. 2048, 8192 - Why powers of 2 ?

- 8192 seems like a good upper bound range to me. Represents a 2 seconds period at 4KHz loop rate. 10000 would be more logical to me. But the max value could be even higher, it is not like there is a risk cranking this too high. Like 60000 or 65535 since that's the max uint16 value.
- `Deprecate MSP_BLACKBOX_CONFIG in favour of MSP2_BLACKBOX_CONFIG (with uint16_t for num and denom)`. That's what I was going to do.

shellixyz, 2019-06-28:
@stronnag I believe this can be closed now ?

closed 2019-06-28 by shellixyz

stronnag, 2019-06-28:
It can, mea culpa