iNavFlight/inav #4059: FC MatekSYS F722-STD, can use S6-s7 for servo, are posible use S5 ?

Papipapito, 2018-12-04:
## Current Behavior
Can  use S5 for a Servo

## Steps to Reproduce

1. Load Profile multirotor
3. Select quad X
4. Add mixer, start in S6, S7, but not S5 

## Expected behavior

When add mixer, add into S5 to use with servo

# version
# INAV/MATEKF722 2.0.1 Oct 31 2018 / 19:27:11 (a04e68aaa)

Regards

shellixyz, 2018-12-04:
It is not possible because S5 shares the timer with S1. You can do it yourself if using the same PWM frequency for motors and servo by building a custom firmware but it won't be added to future releases.

Papipapito, 2018-12-04:
Hello

Tx for the coment, i see this post #3140 , and i search the code of F722 and i see a S8 deactivated with tim5, same of RX2

can use this  s8 for enabling servo? changing the tim_use and buid custom?

 ` { TIM5, IO_TAG(PA3),    TIM_CHANNEL_4, 0,        IOCFG_AF_PP_PD, GPIO_AF2_TIM5, TIM_USE_PPM },`
   ` { TIM3, IO_TAG(PC6),    TIM_CHANNEL_1, TIM_EN,   IOCFG_AF_PP_PD, GPIO_AF2_TIM3, TIM_USE_MC_MOTOR |                    TIM_USE_FW_MOTOR }, //S1 DMA1_ST4 MT1`
    `{ TIM8, IO_TAG(PC7),    TIM_CHANNEL_2, TIM_EN,   IOCFG_AF_PP_PD, GPIO_AF3_TIM8, TIM_USE_MC_MOTOR |                    TIM_USE_FW_SERVO }, //S2 DMA2_ST3 SV3`
   ` { TIM8, IO_TAG(PC8),    TIM_CHANNEL_3, TIM_EN,   IOCFG_AF_PP_PD, GPIO_AF3_TIM8, TIM_USE_MC_MOTOR |                    TIM_USE_FW_SERVO }, //S3 DMA2_ST4 SV4`
   ` { TIM8, IO_TAG(PC9),    TIM_CHANNEL_4, TIM_EN,   IOCFG_AF_PP_PD, GPIO_AF3_TIM8, TIM_USE_MC_MOTOR |                    TIM_USE_FW_SERVO }, //S4 DMA2_ST7 SV5`
    `{ TIM3, IO_TAG(PB1),    TIM_CHANNEL_4, TIM_EN,   IOCFG_AF_PP_PD, GPIO_AF2_TIM3, TIM_USE_MC_MOTOR |                    TIM_USE_FW_MOTOR }, //S5 DMA1_ST2 MT2`
    `{ TIM1, IO_TAG(PA8),    TIM_CHANNEL_1, TIM_EN,   IOCFG_AF_PP_PD, GPIO_AF1_TIM1, TIM_USE_MC_MOTOR | TIM_USE_MC_SERVO | TIM_USE_FW_SERVO }, //S6 DMA2_ST6 SV6`
   ` { TIM4, IO_TAG(PB8),    TIM_CHANNEL_3, TIM_EN,   IOCFG_AF_PP_PD, GPIO_AF2_TIM4, TIM_USE_MC_CHNFW | TIM_USE_MC_SERVO | TIM_USE_FW_SERVO }, //S7 DMA1_ST7`

  `  //{ TIM5, IO_TAG(PA2),    TIM_CHANNEL_3, TIM_EN,   IOCFG_AF_PP_PD, GPIO_AF2_TIM5, TIM_USE_ANY },`

   ` { TIM2, IO_TAG(PA15),   TIM_CHANNEL_1, TIM_EN,   IOCFG_AF_PP_PD, GPIO_AF1_TIM2, TIM_USE_LED}, //2812 STRIP DMA1_ST5`

Maybe I said nonsense, I do not know much about programming

TX

shellixyz, 2018-12-05:
Yes if you don't use the PPM input and UART2 you can comment the first line and add a servo output on PA2 it should work

Papipapito, 2018-12-05:
Tx for help

closed 2018-12-05 by Papipapito