iNavFlight/inav #4062: Autoswitch OSD layout during failsafe

flygemaskin, 2018-12-05:
## Current Behavior
When failsafe occur the current OSD layout stays active.

## Desired Behavior
When failsafe occur autoswitch to a preselected OSD layout with GPS coordinates and other useful information. 

## Suggested Solution
Implement a setting to enable a preselected OSD layout if failsafe occur.

## Who does this impact? Who is this for?
This is probably useful for everyone, especially those of us who enjoy flying with a minimal OSD layout.

dec-dec, 2018-12-05:
YAY! 
This would be a great feature. It´s nice to have a clutter free view with minimal distractions. And great to have that GPS info for when you really need it. Thumbs up and +1

wx4cb, 2018-12-05:
i know right now, when i arm my osd switches, but that's cus i messed up the modes tab lol.. so i know it can be done with FS on a switch, so maybe, if you are sensing failsafe by switch positions being held by the RX you can put an osd switch to that position as well i guess. but it would be nice to have a setting or two...

krzysztofmatula, 2018-12-06:
Not multiplying configuration options maybe just always switching the OSD to layout #0 on failsafe?

Some time ago I proposed more generic approach of automatically selecting layout based on current flight mode - see: #3287.

shellixyz, 2018-12-06:
I propose to add a `failsafe_switch_osd_layout` boolean setting and switch to layout #0 during FS when enabled as suggested by @krzysztofmatula

teckel12, 2018-12-06:
I never seem to have a failsafe to trigger this. Motor lost a mounting bolt and vibrated off, quad crashed and never a failsafe. Maybe if we had a "isFlying" calculation we could also create a proper "has crashed" state which would be super useful to trigger an alternate OSD view.

shellixyz, 2018-12-06:
@teckel12 I don't get it. If you didn't failsafe what prevented you from switching the OSD layout with your RC ? Anyway we need flying detection and it will be implemented at some point.

teckel12, 2018-12-06:
@shellixyz I guess I was considering making something automated where a switch wouldn't be needed. I've been running into count limitations with switches in INAV so I figured automating this would free up a needed slot. It wasn't a problem in the case I gave because I show the GPS coords all the time on the OSD (and I also have them via telemetry).

fiam, 2018-12-06:
I think the best course of action would be adding user-defined dynamic boxes instead of hardcoding specific use cases. It would provide a more generic solution and give users a lot of knobs to play with.

Although this specific use case hasn't been mentioned while discussing dynamic boxes, it could be supported making a user-defined box that's activated via another box/mode. Users could define a box with action = "use OSD layout n", activate when BOXFAILSAFE is active.

This would also be helpful for e.g. making the camera start recording when the craft is armed (add a box with action = "start recording", make it active whenever BOXARM is active).

shellixyz, 2018-12-06:
@fiam You dynamic boxes idea is very good. In the meantime the implementation is very simple for forcing the display of the OSD and switching to the first layout I think we can merge #4064 in case dynamic boxes don't make it in 2.1.

fiam, 2018-12-06:
@shellixyz I'm not against implementing this functionality as a dedicated check in the OSD config, but eventually we should try to move to dynamic boxes (not just for this feature, also for OSD layouts, camera modes, PINIO, in-flight adjustments, etc...).

I don't think it's gonna happen for 2.1 because it requires a significant amount of work, but it should happen eventually :-)

teckel12, 2018-12-06:
@fiam I like your idea a lot. I could even see different OSD layouts for different flight modes. And I agree that the current PR is fine as it is, it can be enhanced later.

PR #4064 (2018-12-09): Add option to switch to layout #0 during failsafe
Closes #4062
closed 2018-12-09 by shellixyz