iNavFlight/inav #4063: Option to output home direction as a data from servo pin so that it could be read with an arduino.

ledimestari, 2018-12-05:
## Current Behavior
As far as I know there is no way to do this now.

## Desired Behavior
What I am attempting create is a freely 360 spinning turntable that would always point home. This way you could use directional antennas on both ends and thus have superior video link.

I have already planned out how to create mechanically moving parts with arduino that would read home_dir and then use continually spinning servo and potentiometer to set it's direction correctly.

I recently created [this system to compansate for roll](https://hackaday.com/2018/12/02/fpv-antenna-leans-into-the-bank/) and from the comments and talk around it, it seems like lot of people might be interested in this type of solution. I would like to take it further from here. 

## Suggested Solution
Add option to output home_dir from a servo pin as a pwm signal for example. 

The scale on which it is outputted doesn't matter much in this case. 0 - 360 or 500 - 2500 or something else is fine, it can be remapped from the device reading it.

## Who does this impact? Who is this for?
Everyone who wants to fly long range with good video. Requires building of the spinning device of course.

## Additional context
I assume resource remapping might be needed to do this, so a feature to remap to free servo slot would be fine. 

I would love to know how to do this myself but I have no experience in modifying flight controller firmware.

I am myself using Matek F405-wing flight controller, MATEKF405SE target, if that matters.

Cheers,
Anders

wx4cb, 2018-12-05:
not so much that you're wanting it in pwm, wouldn't it be easier to have this available in the telemetry? that way you can just use a spare uart talking to the arduino which you're going to be using anyway. this way it would be faster, you can do a whole bunch more stuff, and you're not reading in a pwm value, processing it, then sending it back out as a pwm value?

plus it would have the bonus of being able to have it as a sensor and send that back to the transmitter where a ground station could use the reciprocal value to point its antenna. certainly seems like a useful feature to have it in the telemetry stream.

If that makes sense.

ledimestari, 2018-12-05:
Yes of course uart could be used as well but I thought that you will more likely have a spare servo channel than an uart port. 

You're right about the benefit of having it as a sensor. This would enable one to build very low cost antenna tracker too.

wx4cb, 2018-12-05:
@ledimestari  there's always softserial.

teckel12, 2018-12-05:
@ledimestari Are you wanting this rotating antenna on the craft or the transmitter?

If at the transmitter, you can already calculate craft/home direction with existing GPS and heading telemetry (I do this via a Lua script to show a map as well as home direction on an attitude display). The problem with using this data to control a turntable is that you would need to get the telemetry data from the transmitter to the Arduino, which I can't think of a way of doing that.

If you want the turntable on the craft, it should be controlled via i2c to an ATmega microcontroller (Arduino). The reason you probably want to use i2c is that you wouldn't need a spare UART or any kind of resource remapping. The ATmega/Arduino would drive the servo. As far as calculating direction to home, it would be very easy. Just store the launch GPS location and use the current GPS location and heading to calculate the direction to home. Then, output this via i2c which would be read by the Arduino which would control the servo and rotate the antenna.

The math to do this is simple and I believe it's already being done for the OSD map. Sending this via i2c is all that needs to happen. Well, then the ATmega/servo/antenna thing needs to be built and code written for that too.

In any case, it is doable. But, you'll need to find a programmer who's interested in developing all of this hardware/software. But if you do, I'm sure INAV will be willing to merge a pull request that exports home direction over i2c.

wx4cb, 2018-12-05:
on a side note @teckel12 is that lua script publically available?

as for the other, i never thougt of i2c, good idea. the arduino side of things is simple to do. I was thinking doing the msp side of things and adding it to the telemetry, as that would be easier and less resource intensive on the FC and the arduino side as i want to work on a similar thing. logging telemetry locally on an arduino/sd card would be part of the project i'm experimenting with.

That's why i was wondering if that's already available in the msp stream?

Just starting to learn about all this :D

stronnag, 2018-12-05:
`MSP_COMP_GPS` since forever.

ledimestari, 2018-12-05:
@teckel12 On the craft. 
I2c would also be good way of doing this and I understand the theory behind this that you explained in your comment too but I don't know how to go about making it reality.
I have all the skills to do everything else except what needs to be done on the FC side. One of my original ideas for this turntable was to make it simple enough to be kind of a "plug n play" device that works on it's own and is easy to reproduce for others. 

@wx4cb If this would be useful to you too all the better!

@stronnag I know the values for this probably exist somewhere in the code but how does one output that in readable form?

wx4cb, 2018-12-05:
if i can get the gps data, home dir and so forth from msp i should be good.

teckel12, 2018-12-06:
@wx4cb The Lua script is part of the INAV repo: https://github.com/iNavFlight/LuaTelemetry See the map view for the direction to home calculation. If you have a Horus transmitter see the `horus` branch.

stronnag, 2018-12-06:
> @stronnag I know the values for this probably exist somewhere in the code but how does one output that in readable form?

You send the FC `MSP_COMP_GPS` in with no payload; the FC replies with `MSP_COMP_GPS` out with  a structure in little-endian (for example):
```
struct __attribute__ ((__packed__))  _msp_comp_gps
{
     uint16_t range;
     int16_t direction;
     uint8_t update;
};
typedef struct _msp_comp_gps MSP_COMP_GPS;
```
You format the binary data according to the rules of your implementation language / platform. For `MSP_COMP_GPS` note that you need to normalise the direction, as it output in a +/-180° range.

The MSP spec exists at http://www.multiwii.com/wiki/index.php?title=Multiwii_Serial_Protocol, however you need to check iNav sources as some of the messages have been extended and iNav defines additional messages.

teckel12, 2018-12-06:
@wx4cb It's fairly clear that it currently only works with S.Port, F.Port and D-series telemetry. Crossfire is totally different, but many believe they're the same because Crossfire runs on FrSky. Anyway, compatibility is noted and Crossfire is specifically noted as not compatible as well.

Crossfire is being worked on, but the core problem is that Crossfire doesn't give my telemetry data, so it can't work. That's being changed in INAV 2.1 which will enhance Crossfire telemetry giving required data to function. Soon after INAV 2.1 is released, a new version of Lua Telemetry will support Crossfire. I've been working with TBS on compatibility.

wx4cb, 2018-12-06:
I realise that, but doesnt the data come from the telemetry page of the taranis? If so, the data is available just under different names?

Aside from that, if you want any help testing, im available

closed 2019-02-19 by stale[bot]