iNavFlight/inav #4067: Omnibus F4 Pro V3 - No S6 output resource mapping

500mania, 2018-12-07:
For immediate help, just ask your question on one of the following platforms:

* [Telegram channel](https://t.me/INAVFlight)
* [Facebook group](https://www.facebook.com/groups/INAVOfficial)
* [RC Groups thread](https://www.rcgroups.com/forums/showthread.php?2495732-Cleanflight-iNav-%28navigation-rewrite%29-project)
* [IRC channel](irc://irc.freenode.net/#inavflight)
* [Slack channel](https://publicslack.com/slacks/inavflight/invites/new)

You can also read public documentations or watch video tutorials:

* [Official documentation](https://github.com/iNavFlight/inav/tree/master/docs)
* [Official Wiki](https://github.com/iNavFlight/inav/wiki)
* [Video series by Painless360](https://www.youtube.com/playlist?list=PLYsWjANuAm4qdXEGFSeUhOZ10-H8YTSnH)
* [Video series by Paweł Spychalski](https://www.youtube.com/playlist?list=PLOUQ8o2_nCLloACrA6f1_daCjhqY2x0fB)

500mania, 2018-12-07:
Why I not have the S6 output not mapped to any resource?

[image]

wx4cb, 2018-12-07:
not sure what you're asking

500mania, 2018-12-07:
> not sure what you're asking

[image]

The preset I choos is Aeroplane. all the 6 pm output should be mapped

shellixyz, 2018-12-07:
This is a known issue. It impacts only the output mapping displayed in the configurator, the S6 output will work as expected.

This has already been fixed in the development branch for the next release by #3871. You can use this build in the meantime if you want the output mapping to be displayed correctly: [inav_2.0.1_OMNIBUSF4V3.hex.zip](https://github.com/iNavFlight/inav/files/2657537/inav_2.0.1_OMNIBUSF4V3.hex.zip)

Already fixed, closing.

closed 2018-12-07 by shellixyz

500mania, 2018-12-07:
thanks so much :)