iNavFlight/inav #4073: Omnibus F4 pro v2 (clone) softserial for ESP8266 Wifi module question.

simtcrom, 2018-12-09:
Hello,

In Omnibus F4 pro v2 (clone), I have 3 UARTs and they are used as below,

UART1-IBUS Rx
UART3-Compass SCL/SDA
UART6 -GPS

I see two pads above MOTO_VCC and as per https://github.com/iNavFlight/inav/blob/master/docs/Board%20-%20Omnibus%20F4.md I assume they are CH5 and CH6 used for Softserial.
Can these pads be used to connect ESP8266 Wifi module by enabling softserial in INAV configurator?
As per documentation, it is a "uni-directional port mapped to UART6-TX pin. When enabled, the UART6 is still available as hardware port but it's then RX-only port" Does that mean, GPS connected to UART6 will have issues?

Thanks.

teckel12, 2018-12-09:
@simtcrom If your board looks like that, it's not a pro model. With the pro you connect the battery straight to the FC and it has a current sensor. The pic link you posted is a standard model.

The document you listed is very detailed about soft serial. But you first need to identify the board correctly, which it doesn't appear that you are.

simtcrom, 2018-12-09:
> @simtcrom If your board looks like that, it's not a pro model. With the pro you connect the battery straight to the FC and it has a current sensor. The pic link you posted is a standard model.
>
> The [...]

It is a pro board (but clone, not blue color) for sure. It has current sensor built in and I can connect lipo directly to the FC. I could not find INAV wiki for pro board, I thought PRO board also described in the non-pro page bottom. 
Here is the link to my exact FC, [https://robu.in/product/omnibus-f4-v2-pro-flight-controller/](url)

shellixyz, 2018-12-09:
You can use softserial on CH5 (RX) / CH6 (TX). It is not shared with UART6 on the OMNIBUSF4PRO board. You will probably not be able to use all the baud rates though due to the limitations of the software serial ports.

simtcrom, 2018-12-09:
> You can use softserial on CH5 (RX) / CH6 (TX). It is not shared with UART6 on the OMNIBUSF4PRO board. You will probably not be able to use all the baud rates though due to the limitations of the software serial ports.

I assume by mentioning "OMNIBUSF4PRO " you referred the target and which is for Omnibus F4 Pro.
I am running "OMNIBUSF4PRO_LEDSTRIPM5" which is for Omnibus F4 Pro clones.
Does that make any difference on soft serial?

shellixyz, 2018-12-09:
It doesn't make a difference

simtcrom, 2018-12-09:
I am planning to 

> It doesn't make a difference

My intention of adding wifi module is, for connecting to quad using phone via EZ-GUI for uploading missions, changing settings etc. And also connecting to INAV configurator in windows wireless.
I am planning to follow [http://ez-gui.com/manual/multiwii-clearflight-wifi-to-ezi-gui-how-to/](url)
And planning to buy https://robu.in/product/latest-esp-12f-esp8266-wifi-module-ap-station-remote-serial-wireless-iot-board/
Any idea if it will be compatible?

teckel12, 2018-12-09:
@simtcrom If you have a Pro board, the images and documentation show exactly how to do, specifically:

https://raw.githubusercontent.com/iNavFlight/inav/master/docs/assets/images/omnibusf4pro_ss.jpg

I've used a Bluetooth adapter to communicate with EZ-GUI. That works. Not sure about doing wifi as I've never tried that.

simtcrom, 2018-12-13:
I was able to flash esp-link successfully.
Now softserial is not working. I tested with UART6 and it connect to ez-gui successfully. But not with soft serial.

simtcrom, 2018-12-13:
I changed the baud rate and its working.
Thank you.

closed 2018-12-13 by simtcrom

RichardAnkur, 2019-01-04:
My Bluetooth is set at 11260 baud

simtcrom, 2019-01-05:
> Hello simtcrom
> I have the same flight controller board and I had also ordered it from robu.in
> I have connected my GPS and compass on UART 6 and UART 3 and I am using a flysky fs i6x transmitter and fs iA6B receiver which is connected to sbus/ppm pin. The firmware which I had uploaded is the omnibus f4pro...and not that ledstrip one.
> And now I have connected my HC06 Bluetooth module to UART 1.
> The problem is that when I plug in the battery and connect the bluetooth with EZ GUI app it works perfectly but as soon as I turn on my transmitter.....the female voice from EZ GUI tells data not received.
> Can you suggest me a solution.

I don't know if it matters. In my case,

-  iA6B is connected to UART1 and enabled Serial Rx in ports tab. Also it is configured as IBUS protocol.
- ESP8266 Wifi module is connected to soft serial ports and enabled MSP for soft serial in ports tab.

Attaching screenshot of ports tab,
[image]

RichardAnkur, 2019-01-05:
I can see in inav configurator that there is USB vcp.....UART 1.....uart2.   ....and UART 6 but there is no softserial 1.

RichardAnkur, 2019-01-05:
You GPS is working at 38400 but I for mine baud rate is 11560 and added two 4.7 kohm resistance from scl to vcc and vcc to SDA. Kind of working never tested in for waypoint will there be any issue.

RichardAnkur, 2019-01-05:
For soft serial I should use the two pads above moto_vcc connecting RX of fc to tx of Bluetooth and tx of fc to RX of Bluetooth...... Running GPS(UART 6) at 38400 and softserial at 115600.... That's is what you are suggesting.. Right

simtcrom, 2019-01-05:
> For soft serial I should use the two pads above moto_vcc connecting RX of fc to tx of Bluetooth and tx of fc to RX of Bluetooth...... Running GPS(UART 6) at 38400 and softserial at 115600.... That's is what you are suggesting.. Right

exactly.
But not sure what baud rate you should set for BT module. Try one by one.

RichardAnkur, 2019-01-05:
And also I should use omnibus f4proledstripm5 firmware

simtcrom, 2019-01-05:
> firmware

you can use omnibusf4pro target itself  unless you use led strip.
if led strip needed use the one you mentioned.

simtcrom, 2019-01-05:
> @RichardAnkur The LED strip target is for clones that didn't route the LED strip pin correctly. If you have a non-clone, you don't need to use the LED target at all. I have a few Omnibus F4 Pro FC (non-clones) and I run all with the non-LED strip target and run LEDs on all.

His and mine are clones (non blue colour ones) bought from same hobby shop.

RichardAnkur, 2019-01-05:
@simtcrom are you fully sure that we both have a clone board. You have also bought it from robu.in

simtcrom, 2019-01-05:
> @simtcrom are you fully sure that we both have a clone board. You have also bought it from robu.in
> I also had a doubt but was not sure

I am sure mine is a clone 😀