iNavFlight/inav #4075: Connect iNav via FPort to a mobile device

Schakus, 2018-12-09:
Hey, 
i want to connect my phone for real time mission plannings.
My idea is:
**iNav FC** <-- FPort OR SPort--> **R9 mini RX** <-- telemetry --> **Frsky X-Lite** <-- SPort output --> **Serial Bluetooth Module** <-- Bluetooth --> **mobile Phone/Tablet**

- Could this work with the current features in iNav?
- Is it correct, that i have to use LTM telemetry then?
- Is there a way to realize it?

Thanks!

wx4cb, 2018-12-09:
Dont know about the xlite, but i know it can be done on the x9d

Schakus, 2018-12-09:
Do you have a description how it have to set up on the x9d? I have also an x9d but the r9 long-range system only for the x lite.

wx4cb, 2018-12-09:
Theres plenty of videos on youtu e about it, basically the way i understand it, any of the sensors on the sensors page can be set to come out the serial port

https://youtu.be/DeE1SonEaLw
https://openrcforums.com/forum/viewtopic.php?t=10950

Are just two examples

Bear in mind, i have never done this and any damage to your equipment us not my problem. You do this at your own risk

teckel12, 2018-12-09:
It's also an inverted signal I believe. So you'll need to sort that out either with an inverter or maybe the Arduino soft serial library allows for inversion.

Schakus, 2018-12-10:
Thanks for your answer.

I want to set new waypoints and change some configurations.
This is possible if i connect a Bluetooth module directly to an UART.

It would be great to use the telemetry port for up and downlink and not to use a second communication system.

Do you have informations how it could work? I think its possible with the TBS Crossfire. There is a way to use mavlink.

wx4cb, 2018-12-10:
with crossfire you can using mission planner for android i believe. it's something on my list of things to play with. and if you connect a BT module to a uart, then yea it should work the same. you don't use mavlink, you have MSP on whatever uart you are using.

Schakus, 2018-12-10:
Yes and there is no way to realise it like Crossfire but with Frsky?

wx4cb, 2018-12-10:
not to upload anything TO the aircraft I believe as the frsky stuff doesn't have bi directional capability. that's an FrSky issue.

Schakus, 2018-12-10:
Is there something planned?

shellixyz, 2018-12-10:
FrSky telemetry is bidirectional since it is possible to change the PID settings, VTX power, ... but that's all I know

@wx4cb Do you know if it is possible to use CF's transparent serial bridge with a micro TX and how ?

wx4cb, 2018-12-10:
@shellixyz  I didn't know it was bi-directional, and that makes sense. I never thought about that tbh.

As far as crossfire, not with a micro TX as it doesn't have the bluetooth module in it, whether or not you can add one i don't know. That would be a trappy questions. That being said, they appear in the sensors page, so IMHO, theoretically you can set the serial port on an x9d to output the telemetry as sport and I would ASSUME that would include the crossfire sensors as well. but I don't know

Schakus, 2018-12-10:
Ah yes you're right. I use SPort on my other quads to set up PIDs etc.

The question is, how I could build a test setup. Do you have any suggestions?

teckel12, 2018-12-13:
@wx4cb Crossfire and SBUS telemetry are totally different. There's  few sensors that are the same which makes many believe they're the same when they're not.

FrSky telemetry is bi-directional but with INAV 2.0.1 Crossfire is **not** bidirectional. With that said, I'm working on adding bidirectional support for Crossfire. I have working code already, just cleaning stuff up for a pull request. Maybe it will be in INAV 2.1.

Schakus, 2018-12-13:
@teckel12 I want to use Smartport not SBUS.
And do you know something about how it could work to send configuration data to inav via Smartport? Is it implemented right now?

teckel12, 2018-12-13:
@wx4cb Technically, it's bidirectional as you send flight commands and receive telemetry. But, INAV doesn't currently have MSP bidirectional communication for Crossfire (S.Port does).

Telemetry comes from INAV and there's different telemetry for S.Port and Crossfire. A few sensors are the same, but there's a SIGNIFICANT amount missing from Crossfire. Just take flight modes for example, S.Port has all kinds of info, Crossfire only reports Acro, Horizon, or Angle.

teckel12, 2018-12-13:
@Schakus You can't. SmartPort goes along with SBUS. If you use F.Port that's used for both control and telemetry. I think you may be confused as what SmartPort, SBUS, and F.Port are.

closed 2019-02-26 by stale[bot]