iNavFlight/inav #4076: Cannot connect to Matek F722-SE

amusinov, 2018-12-09:
## Current Behavior

My Matek F722-SE with uart GPS & I2C compass works perfect on Betaflight target MATEKF722-SE. Had several flight sessions today, and decided to try INAV leaving all the wiring as is.

I've flashed this board with INAV 2.0.1 target MATEKF722. Uploaded perfectly, rebooted. 
When I am trying to connect via USB and w/o battery, the configurator cannot establish connection without any messages. 
When I power with battery, the buzzer starts to beep. Connecting USB in addition, the configurator still cannot connect, but it looks like the board is constantly rebooting (serial interface ups and downs).

I've tried standalone macos and crome-based configurator v2.

I guess either MATEKF722 firmware doesn't support SE version, or the firmware cannot load due to unpowered I2C compass.

Would appreciate any help. Cheers, Andrew 

- FC Board name and vendor: Matek F722-SE
- INAV version string: 2.0.1

amusinov, 2018-12-09:
@teckel12 Thanks for explanation, that is great news. Will be waiting for updates. Btw, where can I find dev-2.1 bin?

shellixyz, 2018-12-09:
@amusinov Here is a dev build (3dc3158ac) of the MATEKF722SE target: 
[inav_2.1.0_MATEKF722SE.hex.zip](https://github.com/iNavFlight/inav/files/2660668/inav_2.1.0_MATEKF722SE.hex.zip)

amusinov, 2018-12-09:
@shellixyz many thanks!

amusinov, 2018-12-12:
Hi guys,
have tested 2.1 INAV on MATEKF722SE. Just two issues found.
1) The buzzer doesn't beep at all (wired as and works on BF)
2) RTH mode behaves wierd, flies away and flips.

Decided to put BF back on the quad, but planning to test this firmware again on S800+MATEKF722SE wing soon.

Regards,
Andrew

digitalentity, 2018-12-12:
I'll look into buzzer issue. As for RTH - it's probably compass-related, however it shouldn't flip. Do you have a CLI dump and BB recording of the failed RTH?

amusinov, 2018-12-12:
@digitalentity Привет!
The TS100 compass/GPS had been placed 90 CCW, thus, I've set 270 CW and calibrated properly. Sorry, didn't turn BB on, however, have a few videos from my goggles if could help (will drop some links in 10 minutes). I forgot to mention, in both attempts the quad tried to fly away much over 30 metres I've set, looks like on full throttle.

amusinov, 2018-12-12:
@digitalentity 
https://youtu.be/wZCCkFas3TE

teckel12, 2018-12-13:
@amusinov Honestly, I'd have to guess that the compass orientation is not correct. Maybe it's flipped?  I can't express enough that you need to be **SURE** the the compass orientation is correct before flying. You're testing in the air very high up what I test on the bench without props on.

I have a F722 SE but I'm waiting for the frame and ESCs to complete my build. If you don't sort things out, I should be getting the last parts in the next week. I can also look at the buzzer if digitalentity doesn't get to it first.

amusinov, 2018-12-13:
@teckel12 Hi Tim, good luck with your build.

the compass has been located and declared correctly, I am pretty sure. 
Anyway, I am planning to put F722SE on my S800 fixed wing within next week, and will try to perform testing again.

Btw, guys, are you planning to add Holybro Kakute F7 STM32F745 into the list of supported controllers?

shellixyz, 2018-12-13:
> Btw, guys, are you planning to add Holybro Kakute F7 STM32F745 into the list of supported controllers?

Already done :wink:. See #3868

shellixyz, 2018-12-13:
@amusinov If you'd like to test the dev build for the KAKUTEF7 board: [inav_2.1.0_KAKUTEF7.hex.zip](https://github.com/iNavFlight/inav/files/2676704/inav_2.1.0_KAKUTEF7.hex.zip)

Feedback is always welcome

teckel12, 2018-12-23:
@amusinov The following fixes the beeper and the LED strip:

[inav_2.1.0_MATEKF722SE.hex.zip](https://github.com/iNavFlight/inav/files/2706274/inav_2.1.0_MATEKF722SE.hex.zip)

teckel12, 2018-12-24:
@amusinov I flew the Matek F722SE today with GPS. Position hold and RTH worked perfectly. Tried it several times. Was using the BN-880 GPS correctly orientated.  Also using the development build posted above which fixes both the beeper and LED strip.

I suggest this issue be closed as there's no RTH issue and the other issues resolved via PR.

closed 2018-12-25 by digitalentity

amusinov, 2018-12-25:
@teckel12 I've currently installed inav on Matek F405-ctr on my flying wing. The F722 is currently on my quad with BF. Thank you guys!

teckel12, 2019-01-14:
@MartyFufkin Try the HEX I posted above.  It's slightly before the RC1, but not by much.  I'm running that firmware without any problems.  If it works, I can look at the difference in the target between that firmware and the RC1 firmware you're having a problem with.  I can't test RC1 on my build as I'm in the middle of other testing.  But, I do believe other developers are running RC1 on the F722-SE without issue.

MartyFufkin, 2019-01-14:
Thanks @teckel12 ...  I downloaded and installed (after unsoldering my GPS) and it installed right away without boot loop... so, I then tried the RC1 and it also worked.  I moved GPS to UART3 and all good.  I think the GPS/MAG was the culprit all along.  Thanks for your help.

alfa79, 2019-02-15:
Hello,
I have the Matek f722 se and it will not let me connect in inav flight.
makes me:
No configuration received within 10 seconds, communication failed, Serial port successfully opened with ID: 37.
I try to connect with the battery and I have also tried without battery.
can anybody help me?
Thank you.

teckel12, 2019-02-15:
@alfa79 Not the fault of INAV.  I have the Matek F722SE and it works fine with the configurator.  Sounds like maybe you don't have the correct comm driver installed.  Does the Betaflight configurator communicate correctly?  If you still have a problem, you probably want to open an issue in the configurator repo.  But I can confirm that the Matek F722SE works with INAV and the configurator.

alfa79, 2019-02-15:
thanks for answering. Yes, in the betaflight it works well and in inav with other controllers it also works. but the matek f722 is, does not connect me. I do not know where the problem may be.

MartyFufkin, 2019-02-16:
Alfa79, I had same issue... I fixed it by moving the GPS to UART4 instead of UART2.  UART2 seems to cause issues on this board if a GPSnis connected when trying to connect.  You may be able to test it by depowering the GPS or unsoldering it from UART2 when trying to connect and flash.  At least this was my problem and how I fixed it.

alfa79, 2019-02-16:
Thank you MartyFufkin, I have the controller without connecting to anything, only to the esc 4 in 1. The firmware introduces it to me without problems.

alfa79, 2019-02-16:
Good morning, I have solved it, I have loaded the firmware that put Shellixyz, inav_2.1.0_MATEKF722SE.hex.zip.
It is the only one that works for me.
Thanks to all for the help.
A greeting from Barcelona.

bangkokpete, 2019-05-26:
Hi, having a similar problem, cannot load OSD characters to FC, stops alleyways at 302/5xx. after that no more video out. reset to BF 4.0.2, install default characters works, back to INAV, start-up message invalid font. AND after that I have constant boot-loop. cannot access the FC under INAV anymore. desoldered all Uarts, no difference. Also copying the diff file into CLI does only shows the last line. anyway to complete reset the FC and the OSD chip ? thanks for your support and your work, already tried 3 F722SE and had similar problems. OmnibusF4pro working perfectly. seems to be a F722SE problem.
Peter

damsport11, 2019-06-02:
Same problem here. Matek 722 STD, Flash goes OK, then connect does not work:
No configuration received within 10 seconds, communication failed
Betaflight works fine GPS on TX/RX4, TX/RX1 is free

MartyFufkin, 2019-06-02:
Try unplugging the wire to the GPS (or unsolder one power wire) and try 
again to connect.  This worked for me.  You may have to play with Zadig 
and/or ImpulseRC Driver Fixer to get it working but I did resolve it this way.

HoyDAviDJ, 2019-07-16:
I was having the exact same problems.  I could not get the Matek F722SE into DFU for anything.  Even plugging in USB with the boot button pressed did nothing.  Once I unplugged the GPS module (Matek M8Q-5883) I was able to put the board into DFU directly from Betaflight like normal and was able to flash without issue.  Once I plugged the GPS back in and reconnected it, Betaflight was no longer able to put the board into DFU.  

I am just glad the issue is not unique to me.

DzikuVx, 2019-07-16:
@HoyDAviDJ is the GPS was powered from 5V pads, not 4V5 pads, here is the reason: 

https://www.youtube.com/watch?v=W1vLTbvea3A

HoyDAviDJ, 2019-07-16:
@DzikuVx That wasn't my issue.  My GPS is connected to a 4V5 pad.  I wired the GPS in the same way that it is shown in the F722 SE Manual.  When USB is plugged in, the board boots perfectly fine and connects to Betaflight.  The GPS even gets a 3D fix within about 10 seconds of booting.  

The issue was that when the GPS is connected, Betaflight cannot put the board into DFU with the button on the Configuration Tab or by using commands in the CLI .  The boot button does not put the board into DFU either.  

However, when booting with the GPS unplugged, Betaflight is able to put the board into DFU without issue and flashes normally.  

After I posted my original message I was reading the Betaflight Wiki and found the reason for the problem.  The STM32 group of MCUs look at multiple peripherals, including USB VCP and UARTs, when in Boot Loader mode.  So, if something connected via UART is sending data that looks valid, then the boot loader focuses onto that instead of the USB VCP that it should be.  So, once the GPS is disconnected (at the very least the power to it) then the boot loader focuses onto the correct item, USB VCP, and then Betaflight registers that the device is connected and in DFU.