iNavFlight/inav #4079: furiusFpv F35 with R9M and R9 mm receiver

effysh, 2018-12-12:
## Current Behavior

## Steps to Reproduce

1.
2.
3.
4.

## Expected behavior

## Suggested solution(s)

## Additional context

---

- FC Board name and vendor:
- INAV version string:

teckel12, 2018-12-14:
@effysh There were changes for port inversion between 1.9.1 and 2.0.1.  The R9 MM has both inverted and non-inverted pins.  Then there's the issue of the FC, I assume it has an SBUS port that's inverted.  With the R9 MM, it's best to use a UART that's not inverted on the FC and the non-inverted pins on the R9 MM.

In addition, posting your INAV `dump all` would help further.

effysh, 2018-12-14:
Thanks teckel12
You were right i just did:
smartport_uart_unidir = ON on cli
Telemetry now works

shellixyz, 2019-02-02:
Fixed, closing

closed 2019-02-02 by shellixyz