iNavFlight/inav #4088: R9M and R9MM Failsafes issues

Xedos9er, 2018-12-17:
Hi,
I'm running a R9M and R9MM combo with latest flex 181109 firmware via f.port connected to TX2 pin on my Matek F405 wing (INAV 2.0.1). I have some weird fail-safes during flight even with good rssi. Sometimes I'm not able to abort failsafe even with rssi at 60+, telemetry is still working and my status led on the R9M shows that connection works. Sometimes restarting my taranis qx7 helps, sometimes I just have to wait.
Check to footage:
https://youtu.be/cUBZZq3gaBo

From 2:00 the fail-safe adventure starts. At 7:05 I switched off my taranis, you clearly see rssi drops to 0 and after switching on it gets back to 50 and then I can recover. Are there any known issue with f.port connection? Why does INAV keeps saying "rc Rx link loss" even when connection seems to work (check rssi values).

nmaggioni, 2018-12-18:
I've experienced the same issue with my wing and the same hardware, thankfully I have total faith in iNav's RTH :sweat_smile:

I personally know people who use the R9 system on F.Port without particular problems on their quads running Betaflight, but comparing the two `fport.h/c` files ([1][1], [2][2]) led to no relatable differences - to my eyes, at least! - so I don't really think this could be an iNav issue. The overall condition of the R9's firmwares' stability does also make me think so.

[1]: https://github.com/iNavFlight/inav/blob/56cd0f2c59bd9cf657c4e3362312db7222488a95/src/main/rx/fport.c 
[2]: https://github.com/betaflight/betaflight/blob/dd7a77e46c5c304b251bd1a1bf7fdcfcb47abf68/src/main/rx/fport.c

GhostRiderRC, 2018-12-30:
I think this is a problem with frsky and not inav. Try the stable fw for the r9 stuff, only telemetry on 25mw in lbt but atleast no failssafes.

Xedos9er, 2019-01-11:
Latest r9 firmware (181228 final) did not fix this issue. Somebody told that there still are some timing issues with F.Port and INAV.

nmaggioni, 2019-01-11:
It is now working well for most people (including me): you may have problems with noisy hardware. Being this issue so widespread, I'm positive that it's not depending on iNav/BF's implementation but rather on the TX/RX ones.

You may want to pursue further support from FrSky and help them reproduce the issue - I wasn't able to do so but you may have found a pattern I didn't see.

Xedos9er, 2019-01-11:
I suspect the servos to be noisy. Any information on your hardware an placement of you R9MM ?

Xedos9er, 2019-01-11:
Thanks for the input. Some EMAX 09MD´s are on the way. Maybe I swap them and give it one more chance. 
I could also try to place the R9MM into the front, where the battery is placed recheck for failsafes. If they not occure, I can say for sure, that there´s some servo interference going on, since the servo leads are running parallel to the R9 leads at the moment. I could also test some coax style wire and run the F.port signal on the inner lead and the ground on the shielding of the wire.

If it´s an interference issue, I can tell for sure that F.Port ist much more sensitive, because SBUS version works flawless

yo8aiv, 2019-07-30:
hi.
 i just want to add .. had recently a failsafe  in close proximity <100m after coming in from an 1km flight.
inav 2.2
omnibus F4 v2 pro
Fport on uart1 -  tx1