iNavFlight/inav #4089: Public slack link directs to insecure website

jaxxzer, 2018-12-17:
https://publicslack.com/slacks/inavflight/invites/new

This website does not have a known certificate, my browser tells me I should leave.

wx4cb, 2018-12-17:
looks like the SSL certificate expired today: 

NET::ERR_CERT_DATE_INVALID
Subject: publicslack.com

Issuer: Let's Encrypt Authority X3

Expires on: Dec 17, 2018

Current date: Dec 17, 2018

fiam, 2018-12-17:
We deprecated slack in favor of telegram a few months ago. Where is that link found? We need to remove all references to slack to avoid confusion.

jaxxzer, 2018-12-17:
https://github.com/iNavFlight/inav/blame/master/README.md#L55

fiam, 2018-12-17:
Thanks for the heads up. I was under the impression that it had been removed in the development branch and it would be updated on the next major release, but it's not the case. I'll remove it.