iNavFlight/inav #4091: batt voltage and flytime disappears

yeehaa1, 2018-12-17:
## Current Behavior   omnibus f3 board  inav 2.0 and 2.0.1 when batt volt gets to low it disappears from screen , also after 10 min. flying the flytime disappears from screen

## Steps to Reproduce

1.
2.
3.
4.

## Expected behavior

## Suggested solution(s)

## Additional context

---

- FC Board name and vendor:
- INAV version string:

shellixyz, 2018-12-17:
There is an issue with the OSD chip on your FC. You need to enable `display_force_sw_blink`. Go into the CLI and type `set display_force_sw_blink = ON` and enter then type save and enter.

closed 2018-12-17 by shellixyz

teckel12, 2018-12-18:
@shellixyz Just an FYI. I believe the issue has to do with the VTx or goggles and not the FC. And typically it's only a problem half the time (something to do with an odd or even frame transmission). 

Which is why I always set my builds to sw_blink to avoid issues. I also believe this should be the default as this is a quite common problem and you don't always know it's a problem till you need it.

yeehaa1, 2018-12-18:
the older version inav didn't do it just 2.0  and 2.0.1  and all other info stays on screen      how would frame transmission effect the flytime disappearing at 10 min. markgonna give the fix a try tomorrow to see

shellixyz, 2018-12-18:
@teckel12 Personally I never had the issue. I don't think the issue can come from the VTX it is just passing the signal through. Maybe the Goggles but I doubt it too. It shouldn't be a framing issue because the on/off duration is much higher than the frame rate.

@yeehaa1 Because of alarms making items blink. Default fly time alarm is 10 minutes.

yeehaa1, 2018-12-19:
Thanks that takes care of things just had to set flytime alarm in cli longer because its annoying constantly flashing after 10 min

teckel12, 2018-12-19:
@shellixyz I get different results with different goggles on the same FC (which is a US sourced Omnibus F4 Pro on two different quads). It also only happens about half the time.

I prefer box goggles so they're also cheap and probably not up to the NTSC spec. I'm doing a Matek F722SE build with a Matek VTx so I'll see what happens with the same goggles with a different FC.

yeehaa1, 2018-12-19:
it happened on 3 quads that I upgraded firmware on and they all have diff vtx and diff cameras , they functioned on older inav firmware , I also tried 2 diff goggles and a freestand monitor , so to me its in inav   quad setups or the pre configure for 5" gps quad I used         my inav fixed wing are ok
anyway its works now by turn on sw blink in cli

fiam, 2018-12-19:
If the problem is in the googles/VTX, the reason that software blinking fixes it is because the period of the blinking becomes jittery (HW based is taken care by the MAX7456 internal clock, so it always runs at the same rate). It should be possible to change the period of the HW based blinking so the googles pick it, but we need someone who can easily reproduce the problem and can test possible fixes for us.

digitalentity, 2018-12-19:
I had some shitty goggles where OSD was semi-transparent and blinking because OSD chip was only inserting even fields while goggles displayed either even or odd fields at random. Not anymore.

However in this case I believe it's software blinking.

teckel12, 2018-12-20:
@fiam I have a setup thats very reproducible. It still does seem like sometimes it works just fine, which makes it seem like an even/odd frame thing. If you'd like me to try a few builds, I can test them. OMNIBUSF4PRO target.

yeehaa1, 2018-12-29:
Do you know when firmware for matek f722 SE will be available to use on inav

datamann, 2018-12-29:
@yeehaa1
I think it will be available in next INAV release, not sure when that is though. You can use this https://github.com/iNavFlight/inav/files/2660668/inav_2.1.0_MATEKF722SE.hex.zip on your own risk.