iNavFlight/inav #4092: Virtual pitotmeter

hali9, 2018-12-18:
## Current Behavior
If there is no hardware pitotmeter on board, the air speed can not be estimated.

## Desired Behavior
Since when it is possible estimate the wind, a virtual pitotmeter can be created.

## Suggested Solution
This can be done by subtracting the wind vector from the velocity vector.
We will get the air speed in this way, it is easy to convert into pressure and put it to the driver.

## Who does this impact? Who is this for?
A virtual pitotmeter can be used for example to improve the operation of TURN_ASSIST.

## Additional context
Of course, we convert the air speed to pressure, then in pitotmeter.c again convert the pressure to the speed, but it is probably simpler and clearer than adding a condition in pitotmeter.c.

shellixyz, 2018-12-18:
For now the attitude and wind estimations are not precise enough but that's on the TODO list

teckel12, 2018-12-19:
@shellixyz :thumbsup:

Richrfl, 2018-12-23:
If this feature is implemented, could this airspeed (or at least that provided by a pitot tube, or GPS be used to control the parameter tpa_rate, rather than throttle position?, or at least have an option for the tpa_rate to select the airspeed source?

shellixyz, 2019-01-01:
Yes it is one of the goals. PID attenuation based on airspeed.

KaisarX, 2019-02-03:
I know this is for a virtual Pitot tube... but a semi related and possible solution would be a GPS based wind correction.

Fixed Wing:
step 1.  Gain safe altitude.
step 2.  Set non-stalling throttle (ie 1500/1600us pulse)
step 3.  Allow a moment for airspeed to stabilize
step 4.  perform a 30 degree bank and hold bank, do not try to maintain position
step 5.  mark GPS location, current heading and current time
step 6.  Every time you return to Step 5's heading, mark GPS location and time
repeat step 6 as many times as you like for improved accuracy
step 7.  resume normal flight operations while calculating the difference between GPS locations and time between locations. 

For quads and the ilk
step 2,3 and 4 are replaced with hold altitude and stay level.  Allow the craft to drift.
step 6 is replaced with measure multiple GPS locations and time
 
Result:  you will determine the rate of displacement(drift) per second -> windspeed and direction
apply this vector to any gps ground speed vectors produced to determine actual airpseed.

Windspeed will typically not change over the course of a 30 minute flight in a significant way. 
   with the exception of storms (approaching or departing)  or mountain weather.

Alternative solution... MUCH more difficult to program... potentially...
Calculate where you should be by adding up all headings and an average velocity at a cruise throttle setting, and compare that to where you are.  With enough data and enough time, you will be able to sift out the windspeed and then find an airspeed....   much easier said than done I'd say... but I'd offer it as a challenge.

shellixyz, 2019-02-03:
We already have continuous wind estimation for fixed wing based on GPS and we know it can be made more accurate. It is probably just a matter of improving the attitude estimation. Then we can use it for estimating airspeed. For MRs we don't really need airspeed.

KaisarX, 2019-02-04:
fair enough.  Well done :P  I guess my post can be nixed.

Richrfl, 2019-02-04:
Gents, I fully agree that GPS speed is a good approach to start with. However, I am assuming that what when you make reference to "wind" speed, you mean "air" speed. If this is not the case, please disregard the rest of this message.
If we are talking about air speed, there is a hell of a difference with GPS speed, which is actually ground speed. Good for copters but no for fixed wing. You may have a good ground speed and stall and crash your plane when you have a 10mph (about 15 Km/h), tail wind, or you can land crazy fast and furious with nose wind. This is not considering a 45 degrees or more climb or drop where ground speed and air speed are quite different.
This point is beyond how accurate the GPS calculated speed may be, but about stall, which in a copter is controlled by the motors speed, while in a plane depends on the air speed going through the wing. On top of that, the pitot tube data does not requires the calculation process required to obtain the mobile velocity like the GPS requires.
I still think that to make iNav fixed wing friendly, air speed should be included in the code.
KaisarX, about your suggestion, I do not disagree, but I am not interested in figuring out at what speed I was flying, but at what speed I am flying live. Anyway, I can have my telemetry to give me the GPS speed at anytime. Air speed is important to define stall warnings and to control the aircraft landing.

shellixyz, 2019-02-04:
@Richrfl I was talking about wind speed not air speed. When you have the air speed, attitude and ground speed you can estimate the air speed.

Richrfl, 2019-02-04:
Wow! Absolutly yes, shellixyz!, how do gather information to estimate the wind speed and direction in a moving vehicle? or again we are talking after the fact speed?

shellixyz, 2019-02-04:
The wind estimation is done as described in this paper: [WindEstimation.pdf](https://github.com/iNavFlight/inav/files/2828728/WindEstimation.pdf)

Richrfl, 2019-02-04:
Thank you shellixyz. I'll read that paper.

Richrfl, 2019-02-04:
Shellinxyz, the way described in the paper is really interesting and acceptable. I don't understand for what reason this procedure was implemented, except to avoid the use of another piece of hardware (pitot), and use the GPS data, which is almost a must in the implementation. On the other hand, I think that implementing this method takes a lot more process power than using a pitot. Anyway, I acknowledge the effort and skills of the development team to implement such a complex process into the system.
However, I honestly think (and I am not such a good sw geeck) that adding code to obtain air speed out from ground speed/wind speed, does not make too much sense compared to adding code to use a pitot for air speed, given that as you mentioned before, that air speed has non sense in MCs anyway.

shellixyz, 2019-02-04:
The reason is simply that in a lot of cases adding a pitot is impractical but almost every aircraft running iNav already has a GPS. The processing power needed for the wind estimation is not that big.

Richrfl, 2019-02-04:
Well, that's 100% true. Now, just to update myself in the current status: I understand wind speed and direction is already implemented, but at this time air speed is not there. Is that correct?

Richrfl, 2019-02-04:
On the other hand, air speed sensors are accepted in iNav, so, if air speed is not used, what would be the reason to install a pitot?  I may be off on this one. I am just at re-taking the iNav project today, with a new F7 board, after a few months, and I need to go over all the settings again, therefore, I most probably missed some speed related settings.

Richrfl, 2019-02-04:
Quite understood, Michael. Thank you for your attention and effort. I wish I would have a grade in sw to help.
Nice to have discussed this with you.

KaisarX, 2019-02-06:
Hey, been away, didn't reply.
I"ll define terms quick
Windspeed: The vector of a body of air as it moves in relation to the ground
Groundspeed: the vector of a body as observed in relationship to the ground 
Airspeed: the vector of a body as it moves in relationship to the body of air it travels through.

As such, you can determine windspeed from performing a rate one turn(or a turn at a constant angle would suffice) and observe the distance traveled over time when returning to the same heading.
Using windspeed you can determine airspeed when you only have a groundspeed to calculate from.

The original post was suggesting it was possible to determine airspeed from certain pressures... it was not clear to me what that meant, so I offered a more readily available suggestion.  
I agree that knowing airspeed is key for stall speeds, maneuvering speed, and the like.  And I think it would be possible to set up a procedure for doing a full inflight calibration of all those important and fixed wing friendly numbers.
As a side note, what I might recommend in the interm(aside from writing the code itself)   Is follow this procedure manually.
0.  set up gps and turn on blackbox logs
1.  take your aircraft to a reasonably safe altitude.
2.  set throttle to cruise and level flight   
2.  Put it in angle mode (with a limit of 30 degrees) and put full sticks to one direction for 2 full turns.
3.  go back to level flight and face into the wind.
4.  slowly pull the throttle back.  
         I think this is correct, but INAV should attempt to maintain altitude by adjusting elevator/elevons 
              appropriately, if not don't worry there's a way to correct for it.
5.  at the point the plane can no longer maintain altitude you have your stall speed (and approximate throttle threshold)
6.  Use the blackbox logs to learn your stall speed, but more importantly you now know what is the absolute minimum throttle setting before you can no longer maintain altitude.

Now, using that throttle setting.  
    You probably already know this but stallspeed changes with a multitude of features.
  - angle of bank    - more bank means higher stall speed
  - amount of power  - more power means lower stall speed
    SO, in level flight your stall airspeed will be whatever it was during your test flight which directly correlates to your throttle setting.  But as soon as you turn, or change your attitude, or change your power, your stall speed will change.

Therefore, if you are going to use autonomous flight, i'd recommend setting your cruise throttle to be at least 30% more than your stall throttle (though airspeed does not linearly change with throttle, you might have to experiment)

for example, during a 45 degree turn, your stall speed increases by approx. 18%  So that gives you a margine for safe flying.

OK... maybe i've gone on long enough..
Happy safe flying.

Richrfl, 2019-02-07:
KaisarX, thank you for all the information, and I wish I would be smart enough to add the code in iNav, given that as I said before, I need live data, precisely, to avoid stall. 
Although air speed may be interesting to brag about after a flight, that is not my objective. The key reason I need live data is to safely flight a fixed wing under wind conditions.

KaisarX, 2019-02-07:
Ok fair enough.
I know I gave a convoluted way of describing all the mechanics of flight, and how you might use them.

How large is the craft that you are flying?  Have you flown it before?  How much angle of bank do you want?  Do you want it to be autonomous or semi-autonomous?

What I want to get at, is once you know your "stall throttle"  (which is a totally made up thing that doesn't exist, but really means the amount of power that provides enough airspeed to maintain altitude while flying straight and level)  you can infer a safe setting.  If you are limiting yourself to a 45degree angle of bank, then 40% extra power may be enough to keep you away from stall.  ie. if you fail to maintain altitude at 1300us, i'd say give at LEAST 300 *1.5 +1000= 1450.  Then test and move forward.  

The beautiful thing about airspeed is that if your ability to maintain altitude is good in one direction, it is good in every direction... it's when you increase your angle of bank that you will get into trouble.  Which usually happens when you are trying to maintain a certain location on the ground, and then bank too hard and then run out of airspeed (at 45 degrees stall speed increases by 18%  at 60 degrees the stall speed increases by 40% and at 75 degrees it's 100% and at 90 degrees it's .... well use your rudder instead :P)

When i've flown (i'm a pilot) in high winds, i've found myself tempted to bank hard when I'm trying to hit certain landmarks, or have seen how it's possible to get trapped in an illusion where it 'feels' like you are loosing airspeed, but it's just the groundspeed.  As long as you maintain a sane bank angle and your throttle you are usually fine... and when in doubt reduce bank angle and tip the nose down.  It's ALWAYS tempting when you are losing altitude to pull back harder on the stick.... let go, you gain airspeed, and you start to fly again instead of fall.

But maybe I don't understand your actual situation.  I think there may be a low-tech temporary solution while the high-tech stuff catches up.

Richrfl, 2019-02-07:
KaisarX, the piece of data you just wrote its really good! That is something useful, as you mentioned, for a low tech solution, if the wind factor is taken into consideration. 
Anyway, may be I am going too fast, or dreaming too much, but my objective is implement iNav hardware in all my planes. The first one I will work on is my oldest one, a 2.5 Kg, 1500 mm wing span, which, as best as I was able to measure in very low wind day, has a stall speed of about 50 Km/h. From there on, my other planes are medium size EDFs, with much higher wing loads.

closed 2019-07-10 by hali9