iNavFlight/inav #4093: inav keeps telling me "you need to upgrade firmware before using configuration tab"

howlingj, 2018-12-18:
I am trying to set up a receiver on my omnibus f4 v2.1 but every time i change anything in any one on the tabs when i reboot it is says "you need to upgrade firmware before using (tab i tried to change) tab"
I have flashed the board to the latest available firmware (v2.0.1) for the omnibus f4 pro

rb1205, 2018-12-18:
What version is the configurator, not the firmware. You might be using an old configurator version, try uninstalling it and installing back again. It doesn't auto update.

howlingj, 2018-12-18:
my configurator is version 2.0.0 and i have tried the configurater version 1.9.3 
I beleive version 2.0.0 is the latest

howlingj, 2018-12-18:
[image]

shellixyz, 2019-02-02:
Please try the latest version of the configurator and open a new in the configurator project if you still have an issue:
https://github.com/iNavFlight/inav-configurator/issues/new

closed 2019-02-02 by shellixyz