iNavFlight/inav #4095: FOXEERF405 Target

jaime222, 2018-12-18:
Hi I would like you to add this new target to INAV. 

Thank you!

digitalentity, 2018-12-19:
Are you willing to donate a board to one of the developers for testing the new code?

jaime222, 2018-12-19:
If I send you one board to test the new code, how many time do you estimate you will release an INAV version for that board?

If is something less than 1-2 months I will send it to you today.

Thanks

DieHertz, 2018-12-19:
Is Foxeer advertising this board as iNav-compatible? If so, you should seek their support for adding it into iNav.

It's nice that your are willing to donate a board, but for adding support it's a necessary condition, not a sufficient one.

shellixyz, 2018-12-19:
@jaime222 This board is compatible compatible with iNav. I added the support for the MPU used on this board recently and since Betaflight already got a target written for it, it should be easy to port. We only need a board to be able to test the new target, it is much easier for us this way. It can be done in less than 2 months no problem, should be done in less than 1.

jaime222, 2018-12-19:
nice! then let me know where should I send it, so you can release an INAV version for it as soon as possible :)

shellixyz, 2018-12-19:
@jaime222 Could you connect to Telegram so that I can give you my address ?

shellixyz, 2018-12-19:
https://t.me/INAVFlight

digitalentity, 2018-12-21:
@jaime222 @shellixyz if you didn't arrange anything yet, I will probably get the board supported over Christmas holidays.

shellixyz, 2018-12-21:
Ok :+1:

KnuckleUpFPV, 2018-12-27:
Very excited for this. Vertical mounted in my new nano goblin. Thank you digitalentity.

digitalentity, 2018-12-27:
Targets are in progress. I'm closing this issue

closed 2018-12-27 by digitalentity