iNavFlight/inav #4097: S-PORT HELP

mileslew, 2018-12-19:
hey can you pleeeeeaase add resource remapping. I have a r9 mini and it has an inverted sport pin but i only have one inverted pad available that im using for SBUS. I think the only way i can get s-port to work is if i use soft sereal and resource remapping. If you have any suggestions that would be so appreciated. Thanks!!

giacomo892, 2018-12-19:
Please edit the issue with more infos or I am forced to close this one.

teckel12, 2018-12-20:
@mileslew Resource remapping is out of the question for INAV right now due to the complexities with INAV's target audience that includes fixed wing pilots. Finding "another way" up to and including creating your own custom target should be investigated first.

closed 2019-02-02 by shellixyz