iNavFlight/inav #4105: Problems clearing Navigation Safe flag with no GPS

RichMacf, 2018-12-24:
I have a faulty GPS, and wanted to fly without it, but couldn’t clear the Nav unsafe disable flag.

I disabled GPS in Config, and RTH in Failsafe. The Nav Poshold and RTH modes disappear in the Modes page. But I still get a Nav unsafe flag, which is annoying.

The problem is that the aux entries still exist for the Poshold and RTH modes,  #8 and #9, even if GPS is disabled, and the modes page doesn’t show them. These aux entries seem to throw up the disable flag. It’s confusing that the modes still exist when not visible in the UI.

One solution is to delete the aux entries when GPS is disabled. The other solution is to show the modes in the Modes page if the aux entry exists, so they can be easily deleted.

OMNIBUSF4PRO 2.0.1
Omnibus f4 Pro V3
INAV config 2.0.0

teckel12, 2018-12-24:
@RichMacf What about turning off extra arming safety?

laurentopia, 2019-01-27:
> 
> 
> @RichMacf What about turning off extra arming safety?

how do you do that?

teckel12, 2019-01-27:
@laurentopia Search for `nav_extra_arming_safety`. In CLI, `nav_extra_arming_safety = OFF`

closed 2019-02-02 by shellixyz