iNavFlight/inav #4107: Dynamic looptime for different flight modes

rrk1, 2018-12-24:
Currently. INAV performs all navigation calculations independent of flight mode, which results in sub-optimal loop time. 

We can shut down most navigation calculations while in acro. This will lead to possibility to easily achieve 8K looptime , flight performance on-par with Betaflight, giving the best propwash and turbulence handling. When navigation is necessary (RTH requested for example), looptime can be easily dropped back to safe 1-2K.

This mainly of interest for small racing class multicopters.

teckel12, 2018-12-24:
@rrk1 What about when you switch from Acro to Horizon mode mid-flight?

rrk1, 2018-12-25:
We don't need to shutdown level PID controller. Only turn off heavy navigation math. With BF running on typical F405 processor it is easy to achieve 8K/8K looptime even with accelerometer on.

Though I am not an expert on PID controllers, IMHO if I-term is kept properly reset, we shall not expect large transients if level PID is suddenly turned on. It works open-loop while in acro, anyway.

digitalentity, 2018-12-25:
@rrk1 the most heavy calculations are AHRS and position estimation. You need them to run all the time so the moment you switch to ANGLE or RTH the vechicle knows it's state right away.

rrk1, 2018-12-25:
How about waiting for ~two seconds for estimation to stabilize in althold/level mode and then turn navigation loopback on?

Or may be we shall clean up task scheduler a little bit, giving gyro read/PID/motor writes always a hard realtime at 4K or 8K, and run remaining slow tasks in a cooperative loop in background?

Current multitasking behaves a little weird. I can run gyro at about 4K, with clean 250us looptime, but any single disturbance (like I2C error) seems to desync something, and loop starts to fluctuate around ~390us.

To be honest I have never seen a good diagram explaining how scheduler works in current BF/INAV. 
Except that Boris B picture from some years ago which everyone knows...

teckel12, 2018-12-25:
@rrk1 IMHO, 2 seconds sounds like the difference between a successful return and a destroyed quad. I know I've got myself into bad situations and quickly hit the pos hold switch where a 2 second delay would have been really bad.

digitalentity, 2018-12-26:
As @teckel12 said - 2 seconds is a lot. Also, there is a big difference between two seconds on the ground - which is somewhat enough time for AHRS and position estimator to converge and two seconds of flying in random direction with running motors and in turbulent air - which will probably result in both horizon and position to be totally off.

Re scheduling. We don't have multitasking as in "have two tasks of arbitrary complexity running concurrently without disturbing each other". In INAV and BF task iteration is atomic, it can not be interrupted and new task won't be run until previous task completes it's iteration. Scheduler maintains intervals at which tasks are run.

That's why you can't get 4K looptime + GPS + compass + Navigation. While CPU may be powerful enough to process everythning, tasks reading slow devices (compass) and processing navigation data consume a lot of time **in one chunk** making it impossible to have periodic PID execution at 4kHz.

rrk1, 2019-02-17:
So, no solution for current architecture?

Close this issue now?

teckel12, 2019-02-18:
@rrk1 Once F7 optimizations are complete, the hope is that 8k loop times would be possible even with AHRS and position estimation.  As F7 flight controllers are becoming common, I believe this is a better solution than disabling core to INAV features just to get 8k loop times, possibly generating catastrophic failures.

rrk1, 2019-02-18:
Thank you, closing this.

closed 2019-02-18 by rrk1