iNavFlight/inav #4108: Disable feature(s) by default on F3 processor

teckel12, 2018-12-24:
The F3 processor is running out of memory to compile INAV with everything enabled. Decide on one or more seldomly used or not very useful features that can be disabled by default. The advantage of finding two or more to disable is that one could be enabled and it would still compile.

Starting off the suggestion list...

* OLED display

F3 pilots (if there's any still out there) what could you live without?

wx4cb, 2018-12-25:
CPU based SPI? 
Sonar?
External PWM driver?

Those are at least three i have never used - even on F3 boards

teckel12, 2018-12-27:
Sonar and optical flow are two really good seldomly used features for sure. Not sure how much memory they use, but a good place to look.

I'll create a few builds with these and OLED disabled and see what the result is.

Tygrys-1, 2018-12-27:
I do find the LED strip usefull. If the plane or a drone is to be used LOS without the OSD it is a good indicator of the model status like GPS sats or battery condition. It is also good for knowing the orientation and position of the model in i.e. twilight conditions.

wx4cb, 2018-12-27:
yea i wouldn't turn off LED strip personally. all my birds have leds on of some form.

b14ckyy, 2018-12-27:
Makes sense. 
What about future features? As I remember you plan D-Shot support. This for example could be skipped for F3 targets to save memory for other useful stuff. With lower PID loops it doesn't really make sense and if someone really needs turtle mode, i doubt that those persons would stay with F3 boards.

teckel12, 2018-12-28:
OLED is not LED strip. OLED is a display, not lights. LED strip is not even being considered.

teckel12, 2018-12-29:
Sonar and optical flow are already disabled on the F3 by default. Adding OLED (not LED strip) seems to be the logical next option.

teckel12, 2019-01-03:
@shellixyz I'd be good with all but maybe the pwm servo and pitot?  Give some breathing room for more modern/popular features. And a custom build could always turn a few of these on without running out of memory still.

hali9, 2019-01-03:
There are also not used in a normal flight
//#define USE_DEBUG_TRACE
//#define USE_BOOTLOG
//#define BOOTLOG_DESCRIPTIONS
//#define USE_STATS

And maybe:
#define USE_RCDEVICE
#define NAV_AUTO_MAG_DECLINATION - instead it is mag_declination

And this can be enabled:
#define CLI_MINIMAL_VERBOSITY
#define SKIP_TASK_STATISTICS
#define SKIP_CLI_COMMAND_HELP
#define SKIP_CLI_RESOURCES

teckel12, 2019-01-07:
@digitalentity :+1:

closed 2019-01-07 by teckel12