iNavFlight/inav #4113: failsafe_off_delay problems. When using RTH procedure

kardon18, 2018-12-26:
## Current Behavior
Failsafe stage 2 Procedure has both LAND and RTH to choose from, when selecting a means for landing or returning the model.
In these procedures it has - `failsafe_off_delay` and `failsafe_throttle` to choose from.
Both of these are useful for the LAND procedure. 

But...`failsafe_off_delay` also works when RTH procedure is selected. **And it is also set to 20sec by default**. Which is the maximum allowable time for this function.

The fact that `failsafe_off_delay` operates and has a default value (other than Zero) when the RTH procedure is selected. _Is a very bad idea._
This means, that if a _Signal loss failsafe occurs._ The motors will turn off 20seconds after the failsafe is triggered. 
Which will cause a multirotor to fall to its death, after it just made a climb out, when returning to home.

## Steps to Reproduce
1. Leave `failsafe_off_delay` at default 200 deciseconds. Or set it to any value other than Zero. When 
    failsafe RTH procedure is selected.
2. Take the multirotor out for a flight. Then turn OFF your radio transmitter, to trigger a signal loss failsafe 
    condition. (In case RX failsafe is used to trigger mode selection failsafe)
3. The multirotor will follow it's altitude control mode settings. And proceed to return to home.

4. Then after the `failsafe_off_delay` timer has expired. The flight controller will disarm by TIMEOUT. And 
    the multirotor will plummet to the ground. Which is what will happen if you loose RF link.

## Expected behavior
 Such a function was a left over from past days of Cleanflight. Before the RTH procedure was 
 implemented.
 And should not be part of the RTH procedure. **And only part of the LAND procedure, **in it's current state**.**

## Suggested solution(s)
If this function is left as it is. `failsafe_off_delay` should be ZERO (disabled) by default.

But I do see possibility in having the timer there, when using RTH procedure. For a safety motor cut off.

I believe a good addition to the code for `failsafe_off_delay` RTH procedure. Would be for the motor off timer to only commence after the barometric altitude drops below a preset value, that is above the armed home launch altitude.

And I also think its maximum configurable value should be higher than 20 seconds. Maybe 60seconds.

## Additional context

[image]

kardon18, 2018-12-27:
Has anyone else experienced this? Or can anyone confirm that this is a problem. 
I find it hard to believe that this has been over looked.

I have tested the quad on the ground after setting `failsafe_off_delay` to zero. At it doesn't disarm now.

I have looked at the DIFF and DUMP posts of many setups. And most leave their `failsafe_off_delay` at the 20 sec default.

The only other thing I can think of...  Is that most people don't use the valid pulse range, to trigger Failsafe. 
But instead use their _Receivers failsafe_ to instruct the _Failsafe Mode switch selection_ to initiate Return To Home.
Under this situation, the `failsafe_off_delay` should be negated.

Here is a picture of the free fall, before it hit the deck. To show it was Disarm by TIMEOUT. 
Logs were lost. The crash was a bit brutal from 107meters up.

[image]

kardon18, 2018-12-27:
Has anyone else experienced this? Or can anyone confirm that this is a problem. 
I find it hard to believe that this has been over looked.

I have tested the quad on the ground after setting `failsafe_off_delay` to zero. And it doesn't disarm now.

I have looked at the DIFF and DUMP posts of many setups. And most leave their `failsafe_off_delay` at the 20 sec default.

The only other thing I can think of...  Is that most people don't use the valid pulse range, to trigger Failsafe. 
But instead use their _Receivers failsafe_ to instruct the _Failsafe Mode switch selection_ to initiate Return To Home.
Under this situation, the `failsafe_off_delay` should be negated.

Here is a picture of the free fall, before it hit the deck. To show it was Disarm by TIMEOUT. 
Logs were lost. The crash was a bit brutal from 107meters up.

[image]

wx4cb, 2018-12-27:
mine is whatever is default and i hit RTH on the switches the other day and certainly took longer than 20 seconds to get home. that being said, that wasn't technically a failsafe so i don't think that this setting applies.

kardon18, 2018-12-27:
@wx4cb Thanks for your comment.

I usually set `failsafe_off_delay` to zero. But I forgot on this quad.

And I too have had this quad Return to Home with the failsafe test switch. And it flew back for well over 20 seconds.
But this was the first time I tested it with a signal loss (outside valid pulse range) failsafe. By turning the transmitter OFF.

kardon18, 2018-12-27:
@wx4cb Thanks for your comment.

I usually set failsafe_off_delay to zero. But I forgot on this quad.

And I too have had this quad Return to Home with the failsafe test switch. And it flew back for well over 20 seconds, and landed.
But this was the first time I tested it with a signal loss (outside valid pulse range) failsafe. By turning the transmitter OFF.

rb1205, 2018-12-27:
Are you sure about this? Because I'm quite sure I've triggered failsafe RTH for more than 20 seconds before, and my RX is setup to output no data on link loss so that would qualify as regular failsafe.  I always assumed that `failsafe_off_delay` only applies to LAND failsafe policy.

I'll test it and report back.

fiam, 2018-12-27:
`failsafe_off_delay` only applies when failsafe is set to land immediately or when RTH fails and goes into emergency landing. Check the code in `failsafe.c`

kardon18, 2018-12-27:
@rb1205 @fiam 
Thanks guys for your replys and testing. I'm grateful. 
I spent more time over the last few hours with the quad. And I agree with what you are both saying; and how it works.

I have now found that something either got corrupted in the firmware or settings of the flight controller.

One other thing that was acting up as well. Was the GPS symbol on the OSD would continue to flash. Even after the 6 satellites were attained. It **sometimes** took as many as 9 satellites before it would even let me arm the quad. But it didn't do this all the time....Only occasionally did I see this over the last couple of weeks, during this build.

So I decided to re-flash the FC and reload the settings again.
And after doing many ground tests. It seems fine on all accounts now. : /

I have occasionally had setting glitches with other FC firmware's over the years. But this one is more of a   
concern to me, that a corruption in the settings can cause multiply problems at this level.

Kacmarik, 2020-12-03:
Yes I have experienced it, which wrecked my plane!  How is 20 seconds to kill motors after return to home is initiated EVER a good idea.   If you are more than 20 seconds away and have enough power to return to launch and and loiter for 10 minutes why in the world would you kill the motors???  The only reason you would ever want to kill the motors in a RTH scenario is if for some reason during RTH you were going the wrong way.  Is this not obvious to everyone?  A tree gave it's life over this:  https://www.youtube.com/watch?v=PCX8dEEBjEU

Kacmarik, 2020-12-03:
Ladies and gentlemen this is your captain speaking, we have experienced a minor radio reception issue and as a result our auto pilot has initiated the return to home airport sequence.  Although we have enough fuel on board to fly from here to Ecuador and back, I regret to inform you that the default value of failsafe_off_delay=200 will kill our engines in 20 seconds and after which we will plummet to the earth killing everyone on board.  So with the remaining time we have left I would just like to say thank you for flying on Twilight Zone Airlines and ask that you take what little time we have left to ponder why anyone would configure a distressed aircraft to kill it's engines within 20 seconds of a communications failure.