iNavFlight/inav #4124: Advice on what RangeFinder to use with UART

Iamaman00, 2018-12-27:
Hello I am quite new in building a quadcopter, as the title specify what rangefinder I can use for UART connection, as I am using HGLRC F4 Flame V2 Flight Controller which only have 3 UART pads, 1 extra motor pad and no I2C. So I am quite limited in my choices, I do have a lidar lite v3 but having no way of integrating to this FC no extra PWM pads and no I2c. Thank You for any advice.

shellixyz, 2019-01-02:
I guess you could use a Benewake TFMini. Support has been added relatively recently for it. I didn't try it myself though

Stealthwindz, 2019-01-03:
Thank you for the advice, I might buy and try it out soon. I have change the Fc to Airbot OMNIBUS Fireworks V2, tried the altitude hold but not that stable as I am just using the barometer for the altihold. Hopefully if i really buy the TF mini will be helpfull in altihold.