iNavFlight/inav #4125: BMP180 baro bug

spleenware, 2018-12-27:
## Current Behavior
Have a video here: https://www.youtube.com/watch?v=hSrvqG_GzfA
The ALTHOLD and RTH seem to get a punch out at random point in time.
I downgraded to 1.8.3 (from 2.0 RC3, with external Baro code compiled in) and still get the punch out problem. I then downgraded to ver 1.6 and the problem went away. So, external Baro (maybe BMP180 related) bug seems have been introduced at some point.

## Steps to Reproduce
1.  Install any F3 target
2. Configure with BMP180 baro option.
3. Enable ALTHOLD mode
4. 

## Expected behavior
Normal altitude hold

## Suggested solution(s)
Some investigation into external baro (BMP180) support, and what has changed since ver 1.6.

---

(this is the version I have it working on)
# version
# INAV/OMNIBUS 1.6.1 Mar  9 2017 / 00:43:35 (6c878ae)

(Version 1.8.3 and higher is what seems to have the bug)

digitalentity, 2018-12-27:
Do you have a blackbox log?

wx4cb, 2018-12-27:
also, you are covering your baro on the quad?

spleenware, 2018-12-27:
> also, you are covering your baro on the quad?

Yes, have it covered with porous foam.

spleenware, 2018-12-27:
> Do you have a blackbox log?

No. I'll try to capture a session soon. I actually have TWO quads setup now, and both exhibit the same behaviour. One is an Omnibus F3 nano, the other an F4 based FC. 

I thought ver 1.6 was OK, but on a second flight it did the punch out on enabling ALTHOLD. :-(

So, I've tried different Inav versions, and two different quad builds, swapping the baro and GPS around, and they all do the punch-out. (eventually)

Not sure what the common element is. Am suspecting it's maybe something to do with the 'wizard' in the configurator. I chose the '5 inch racer' to initialise the settings after flashing. Maybe there's some nav/alt related PID setting that's too extreme(?)

digitalentity, 2018-12-27:
Likely duplicate of https://github.com/iNavFlight/inav/issues/461 and #486

closed 2019-04-17 by stale[bot]