iNavFlight/inav #4126: DSHOT not functional on Matek F722

DzikuVx, 2018-12-27:
## Current Behavior

## Steps to Reproduce

1.
2.
3.
4.

## Expected behavior

## Suggested solution(s)

## Additional context

---

- FC Board name and vendor:
- INAV version string:

DzikuVx, 2018-12-27:
After further investigation, when DSHOT is enabled, LED is interfering with BEEPER. Disabling LED solves BEEPER but still only S1 output is functional.

digitalentity, 2018-12-27:
Blocks #4111

DzikuVx, 2018-12-27:
Setting 
```
DEF_TIM(TIM3, CH1, PC6, TIM_USE_MC_MOTOR | TIM_USE_FW_MOTOR, 0, 0),   // S1 DMA1_ST4
    DEF_TIM(TIM8, CH2, PC7, TIM_USE_MC_MOTOR | TIM_USE_FW_SERVO, 0, 0),   // S2 DMA2_ST3
    DEF_TIM(TIM8, CH3, PC8, TIM_USE_MC_MOTOR | TIM_USE_FW_SERVO, 0, 0),   // S3 DMA2_ST2
    DEF_TIM(TIM8, CH4, PC9, TIM_USE_MC_MOTOR | TIM_USE_FW_SERVO, 0, 0),   // S4 DMA2_ST7
```

causes board to not boot after DSHOT protocol is selected

closed 2018-12-27 by digitalentity