iNavFlight/inav #4129: Stabilized Yaw for Drag Rudder in Flying Wings

chtay, 2018-12-28:
Hi

I not sure whether current INav servo mixing can have these kind of mixing or not, if not, I hope it can be the next feature instead.

In airplane mode, currently, the "Stabilized Yaw" is like aileron or elevator style, that is swing left and right or top to bottom.
Will it be possible to have an option to swing only half cycle? Meaning, from center to bottom or from center to top only?

Reason of doing so, is for drag rudder (or split rudder) in flying wing. We want the servo that control the drag rudder to activate/move ONLY half of the servo range. It will ignore the other half of the range.

E.g. There is a servo (or two servo working in tandem), to control the openning/closing of the split rudder per wing. When moving rudder stick to the left, the left wing drag rudder will open, while the right wing split rudder remain closed (not moving). When doing right rudder stick, the right wing split rudder open while the left wing split rudder remain closed (not moving).

Currently, this can be achieved by setting each drag rudder servo to be only half the servo range (in the servo tab). E.g. 1000 (min), 1500 (mid), 1500 (max) or 1500 (min), 1500 (mid), 2000 (max).

But by doing so (restrict servo to half, in the servo tab), the drag rudder, now become purely for drag rudder. We cannot use/mix it with aileron or elevator instead. Meaning, we cannot use drag rudder as elevon as well (That is, do not have full servo swing).

rb1205, 2019-01-02:
Looks like something's coming up: #4144

DzikuVx, 2019-01-03:
No, this is not a solution yet.

DzikuVx, 2019-01-03:
I will do something about it for INAV 2.2

wojciechwas, 2019-01-28:
I did this manually in the past with inav 1.6, I am sure you can replicate it now in the 2.1. 
Why don't you set the split so it is half open at mid range and fully closed at one extreme? You can then use the full servo throw. 
Furthermore, you can add a mix for air braking that will be mapped to a slider or a 3 position switch. That is what I used in the past so for landings I could fully open servo and still have yaw authority. :)

DzikuVx, 2019-02-27:
@chtay next release of INAV will have + and - roll, pitch and yaw stabilized outputs that will solve the problem of drag rudders

closed 2019-02-27 by DzikuVx