iNavFlight/inav #4131: Changing mag I2C address

Cecemel20, 2018-12-29:
Hi,
The mag of my gps/mag module isn't working (mag not recognized by iNav). After hours of trying and researching, I connected the mag to an Arduino and ran an I2C scanner sketch. It told me that a device was found at address 0x0e. How can I set the mag's address in iNav to 0x0e?

FC is a Matek F405-ctr

Thanks

Cecemel20, 2018-12-29:
@teckel12 You mean the Atmel732?

Cecemel20, 2018-12-29:
But how could I get iNav to communicate with it?

Cecemel20, 2018-12-29:
@digitalentity Doesn’t the MAG3110 have the same i2c address?

teckel12, 2018-12-29:
@Cecemel20 There's two issues. First, the address needs to match. But also, the packets transmitted need to be of a known format. So, even of if the address matches, it still probably won't work. 

Here's a list of the magnatometers that have a know format and are therefore compatible:

https://github.com/iNavFlight/inav/blob/development/src/main/sensors/compass.h

Cecemel20, 2018-12-29:
@teckel12 I guess the problem is that this board was actually listed for the pixhawk/pixracer. Is there anything I could do apart from getting a new one?

teckel12, 2018-12-30:
@Cecemel20 What does pixhawk compatibility mean to INAV? I would think you would buy something compatible with INAV, not pixhawk.  Something proven like the BN-880.

digitalentity, 2018-12-30:
The chip looks pin-to-pin compatible with HMC5883 or QMC5883, but apparently it isn't since it has some odd "7280" marking. If we know the model of the chip - we could possibly add support for it.

Cecemel20, 2018-12-30:
I don’t really understand.. The [datasheet](http://www.ti.com/lit/ds/symlink/ads7279.pdf) says that it’s a (dual channel?) ADC, so there should be some kind of mag with analog output connected to it?

Cecemel20, 2018-12-30:
@digitalentity Is there anything I could do to help adding support? I have a scope so I could probe some stuff if it’s of use.
I’ll probably buy another module (flashable m8n+LIS3MDL).

Cecemel20, 2018-12-30:
@teckel12 There’s nothing on the other side, except if it’s possible to put components underneath the antenna?

teckel12, 2018-12-30:
On second thought, the 7280 probably does output i2c.  So it's possible that the address can just change and it will work. What target are you using?

Cecemel20, 2018-12-30:
@teckel12 Matek F405-ctr

Cecemel20, 2018-12-30:
@teckel12 I loaded an [Arduino sketch](https://www.sparkfun.com/tutorials/301) which reads data from the HMC5883L and changed the address from 0x1e to 0x0e. No success.. I'll attach a screenshot of the scope.
Maybe I could 'bruteforce' a way to get an output. Just by trying to send a bunch of different data?
Why do you think the 7280 outputs I2C? I don't see it mentioned in the datasheet.
Do you have any idea of what the Atmel ic is doing?
[image]

teckel12, 2018-12-31:
@Cecemel20 The data sheet for the 7280 says serial, which can mean i2c. And, the Amtel chip is for i2c data storage. So, it seems logical that the 7280 outputs i2c.

Do you have the GPS connected to a UART or does that also pass over i2c? I believe the pixhawk GPS is connected via UART, but wanted to be sure in this case.

Cecemel20, 2018-12-31:
@teckel12 Gps connected via UART. The I2C lines are indeed connected to the 7280, but the pins don’t match with the datasheet..
[image]

teckel12, 2018-12-31:
@Cecemel20 First, I'd try the MAG3110 as that already uses the address 0x0E so it may be a clone of that chip.  I've also attached a special build of INAV that changes all of the I2C addresses to 0x0E.  If it's a clone of another chip, it's possible this build will work:

[inav_2.1.0_MATEKF405.hex.zip](https://github.com/iNavFlight/inav/files/2718962/inav_2.1.0_MATEKF405.hex.zip)

If the attached build doesn't work, it's likely that it would need custom support for it.  My suggestion would be to just buy a $17 BN-880, as adding support could take longer than shipping a known working GPS/mag board.  But, if adding support is the only option and you can wait, digging up a spec sheet with more details would be a first start.

Cecemel20, 2018-12-31:
@teckel12 Thanks a lot. I actually already ordered a compatible module yesterday but it would of course still be nice to get this one working and supported. I’ll report once I’ve tested with your firmware.

Cecemel20, 2019-01-01:
One more question (Tell me if this is a more complicated thing, I’ll open a new issue for it then):
Gps refresh rate is 5hz but I thought iNav gps auto configuration sets this to 10hz on m8n module?

teckel12, 2019-01-01:
@Cecemel20 Set to UBLOX7 for 10Hz GPS.

Cecemel20, 2019-01-01:
@teckel12 What do I need to set to UBLOX7?

Cecemel20, 2019-01-01:
@teckel12 Also, will you be adding this mag so it’s supported in future updates? I actually used the matek f405 servoS6 firmware, for camera tilt on S6 but I can fly without it until the next update, not a big deal.

teckel12, 2019-01-01:
@Cecemel20 UBLOX7 is an option just like UBLOX is. It's even in the configurator.

Cecemel20, 2019-01-02:
@teckel12 Thanks, I’ll try it.

I’m having trouble calibrating my mag.. I run through the calibration, after the 30 seconds it says calibration successful but the values still say 0, I press save, go back to first page but there’s still a red cross next to ‘compass calibrated’.

I’m not sure I entered the orientation of my fc/mag correctly, could this be the cause? Please have a look at the attached image. Right now I set ‘board alignment yaw’ to 90 degrees and selected 0 degrees CW flip for the magnetometer (because the mag is on the bottom of the gps module I guess).

Something I noticed: When looking at the mag data in sensors tab, tilting the quad makes the mag values change quite a bit, but ‘yawing’ it (the way I would think a mag is used for navigation) doesn’t really change the values. Is this normal?
[image]

teckel12, 2019-01-02:
@Cecemel20 v2.1.0 is fairly close to being released. So unless another developer picks this up, I probably won't have time, so it may not happen for 2.1.0.

I'd be happy, however, to make a custom build once 2.1.0 is released. This isn't a difficult feature to add, so I'd have time to add this in the next release.

teckel12, 2019-01-02:
@Cecemel20 That's not normal for the mag, it means the orientation is wrong.

Cecemel20, 2019-01-02:
@teckel12 So by selecting the correct mag orientation, this should be fixed? Doesn’t the sensor tab show raw sensor data?

teckel12, 2019-01-02:
@Cecemel20 You should be able to point the quad in 4 known directions N S E W, and not only should it show the correct direction, it should stay in that direction. If it drifts after a couple seconds, it means the orientation of the mag sensor is wrong. It sounds like this would happen in your case. If you're not sure of the orientation, try each one (there's only 8) till it shows the correct direction, and doesn't drift.

Cecemel20, 2019-01-02:
@teckel12 I'll send the results. Could you tell me which one is the correct orientation?
[image]
[image]

teckel12, 2019-01-02:
@Cecemel20 I wouldn't know as I don't have your sensor. You should be looking at the direction in degrees, not the raw data. Set an orientation, look at the direction I degrees in the configurator, point the model in each direction, note I'd it's correct and doesn't drift, if it's not right or drifts, try the next orientation.  Repeat till you have the correct orientation.

Cecemel20, 2019-01-02:
@teckel12 Oww.. hahah. Thanks, I’ll try it.

Cecemel20, 2019-01-02:
@teckel12 No matter which one I pick, the heading value always drifts back to:
120
29
300
210
62
332
241
151
degrees

Cecemel20, 2019-01-02:
Now it still drifts, but the value it drifts to changes depending on the quad's direction. Haven't changed any settings, just a reboot.

teckel12, 2019-01-02:
Have you tried all the flip options too?

teckel12, 2019-01-02:
Also, be sure you don't have a magnetic field causing a problem, like a computer.

Cecemel20, 2019-01-02:
@teckel12 Yup.. tried all 8 options. I’m using a wireless link so my laptop was quite far away (as well as any other metal objects).

Just wondering, if the heading value is drifting, why would changing the mag orientation stop that? Doesn’t it just offset the heading?

Maybe this mag isn’t a clone of the IST8310 and it’s returning different values? I don’t know, just a guess.

Cecemel20, 2019-01-05:
@teckel12 My new module should arrive on monday, I’m expecting it to work..
Maybe you or someone else should add this mag but mark it as experimental? So we may get some feedback from other people?

Cecemel20, 2019-01-05:
@teckel12 Okay.. no need to add as experimental. I tried calibrating the mag again (didn’t change any settings, just plugged in and calibrated) and it worked. Heading is shown correctly. I’m happy and confused..

Board orientation: 90
Mag orientation: 270 cw flip

Doing a testflight tomorrow.

teckel12, 2019-01-05:
@Cecemel20 Awesome! Mag orientation can sometimes seem impossible, I've been there too.

Let me know how the flight goes, and I'll make a note to put this in as a new sensor.

Cecemel20, 2019-01-05:
@teckel12 Yep, I was planning on doing that.

I think that my mag isn’t exactly pointing forward, it’s probably off by a few degrees, as i had to glue it on. Does the calibration fix this?

teckel12, 2019-01-05:
@Cecemel20 Probably not, but it won't matter much as far as the magnatometer goes.

teckel12, 2019-01-06:
@Cecemel20 There's a tail-first RTH that I like to use. I like it because I typically turn RTH off once it gets close. And it makes I easy to fly by sight knowing the orientation.

Cecemel20, 2019-01-06:
@teckel12 Yea, I know. I’m gonna enable it. Do you know whether this mag is going to make 2.1.0? Don’t know how close it is to being released. If not, once it’s out, I might try to figure out how to build my own 2.1.0 with my mag and the servoS6.

teckel12, 2019-01-06:
@Cecemel20 I keep hearing 2.1.0 will be released any day now. Because of that, I'd rather not introduce a new sensor at this point, possibly delaying things. Instead, I'll make you a release version of 2.1.0 with thr new mag to test, then submit it for 2.1.1

Cecemel20, 2019-01-06:
@teckel12 Okay, thanks!

Cecemel20, 2019-01-06:
@teckel12 One more thing: I just tried to fly a mission, planned using mission control, saved to eeprom. Turning on wp mode didn't do anything. Poshold, Althold and rth work perfectly, got 17 sats. The mission is saved because I can load from eeprom and the waypoints appear.

shellixyz, 2019-01-06:
It could

teckel12, 2019-01-07:
@Cecemel20 You may want to open a new issue as well.

Cecemel20, 2019-01-07:
Here you go: #4165

Cecemel20, 2019-01-09:
@teckel12 Would you mind building me a firmware based on 2.1.0 and servo s6 when you've got time?

edwinvp, 2019-05-24:
I can confirm (based on my own experience) that the chip marked 7280 is indeed the IST8310 chip. It shares the same 0xE "7-bit" I2C address with the MAG3110 but it is definitely not digitally compatible with it (nor with the hmc5843).