iNavFlight/inav #4132: Ading manual trimming of an aircraft

Halfduplex85, 2018-12-30:
## Current Behavior

## Desired Behavior

## Suggested Solution

wojciechwas, 2019-01-04:
This can be easily done through radio with OpenTX.  

1. Go to flight modes tab in OpentX. 
2. Assign different flight mode to each of the positions of a switch you use for modes in Inav.
3. Use different trim position for manual mode especially - so when you trim in manual it does not affect trim position for any other flight modes. 

Personally I remove trims from 3D, althold or auto modes and only allow trim for 2D (horison/acro) and manual.

shellixyz, 2019-01-04:
You shouldn't use trims on the radio even if you're using them only for the manual mode. A plane will fly better if it is trimmed in all flight modes.

ShawnzAZ, 2019-01-06:
I wonder if something could be added in the adjustments area to do this, much like we can currently do for board angle.

Artyum, 2019-05-28:
I would like to introduce an idea of extension for servo autotrim feature. The Autotrim in my opinion is generally working but is often painful to get good results. It is also required to land and disarm to save values that is annoying if you'd like to make just small adjustments.

The idea is to implement new on/off switch for example "trim mode". If it's enabled than plane could be trimmed while flying in manual mode with radio's micro-switches just like "the old times :). When trimming is ready and satisfying all sticks should be released (rud,ele,ail) and then disable the trim mode. After that Inav could save new mid position for steering servos.

The trimming would be possible only active manual mode as only in this mode the trim micro-switches can be used. With that approach every mode - manual and stabilised would have the same trim setup. It would be possible to set trim mode always on, just have to remember to release sticks when switching manual mode off. To make small adjustments it would be required for example to enter manual mode, trim slightly with micro-switches and again enter stabilized mode.

FPVZaphod, 2020-03-27:
Did not  see this - opened a very similar request a while ago:
https://github.com/iNavFlight/inav/issues/5033

asgarPL, 2020-10-30:
Recently I've done my INAV MAIDEN manual mode flight and I've to confirm that it is sometimes hard to receive 
satisfying the auto-trim results, without "out of los" fpv assistance. I was trying three times in one flying session and still not happy with its results. Since auto-trimming didn't resulted in "almost" straight flight, so in third try I was even adding some small plane rotation to better tune average servo positions. I know that this thread corresponds to supporting of TX trim switches. But I'm wondering why not to make this more smart and why not to take I-term factors to servos mid-point calculation. I don't know whether it is possible, but maybe it would make this procedure more easier and accurate. Finally, maybe TX trimm switches support actually wouldn't be needed?

FPVZaphod, 2020-10-30:
I stopped using Autotrim completely. 
Default PIDs, but FF raised until control surface throw in acro reaches around 80-90% of the throw in manual. 
This is always fine for a maiden in Autolaunch, and flight in acro mode afterwards. 
Next step is levelling the plane out, and enable AutoTrim. Once done, I test it in manual. If manual flies well, I do some full stick rolls and loops to have the roll and pitch rate on camera. 
Then I land, calculate roll and pitch rate, and enter it to iNav. Then controlling if adjustment of the pushrods is required. If not, start again in autolaunch, and inflight tuning of P and I .  Cruise speed, and rising it until it flies well, without oscillations. 
Then fullspeed-test. If there are oscillations, I raise TPA until it's gone.  
But I need to say, I always maiden with bungee launch. This ensures enough speed to have good control when it leaves the rubber. Bungee Launch is much more safe, there is nearly nothing what can go wrong.

closed 2023-09-29 by DzikuVx