iNavFlight/inav #4135: OSD turn coordinator

GhostRiderRC, 2018-12-30:
Hello,
Would it be possible and is someone willing to add a turn coordinator to the OSD?
Just like a yaw string on a cockpit canopy but digital for when there is no elegant solution or on a mc.

https://en.wikipedia.org/wiki/Turn_and_slip_indicator

GhostRiderRC, 2019-01-03:
So the coordinated turn makes the acc think it's flying level and therefore it is impossible (without an extra gyro canted 30º) to meassure? 
Is it possible to use the error (where the acc thinks it's level) to determine if the turn is coordinated?

I have a glider with the cam inside the cockpit and a yaw string. I have turn assist enabled when I use the rudder stick to pan the cam and it helps somewhat but not enough. I don't have an airspeed indicator.

On gliders it makes a big difference in efficiency/maintaining altitude on multicopters it just looks cool.

shellixyz, 2019-05-10:
@GhostRiderRC In 2.2 you will have the possibility to display lateral g-force on the OSD. I think you can use this to make coordinated turns manually.

Yes there is an issue currently with the artificial horizon going to 0 roll when doing coordinated turns. We're trying to solve the issue.

PR #4696 (2019-05-12): g-force OSD item and stats
Take over from #3566. Closes #3511. Closes #4135 

Matching configurator update: https://github.com/iNavFlight/inav-configurator/pull/765
closed 2024-02-02 by DzikuVx