iNavFlight/inav #4137: 2.1 Download?

CradyLab, 2019-01-01:
Paweł Spychalski indicates 2.1 is available for download:
https://github.com/iNavFlight/inav/wiki/2.1.0-Release-Notes

But the link provided in "Upgrading from 2.0.0 Be sure to run the latest INAV 2.1 Configuration release. Download from here" only shows 2.0.0 as the latest available release: 
https://github.com/iNavFlight/inav-configurator/releases

Is 2.1 actually available and if so where can I find it.  Thanks!

closed 2019-01-01 by giacomo892

teckel12, 2019-01-01:
@CradyLab The document is just in preparation for 2.1.0 release. When released it will be listed on releases. On Github, wiki docs are not versioned along with the repo release, so they need to be written as if the release already exists.

This issue was closed as there isn't an issue, just a misunderstanding of Github.

CradyLab, 2019-01-01:
Thanks for the clarity rather than just cutting me off.  Much appreciated