iNavFlight/inav #4138: VL53L0X distance sensor is not working with OmniNXT F7 FC

SaschaIoT, 2019-01-01:
## Current Behavior
The VL53L0X distance sensor is not working with the OmniNXT F7 FC, because I2C communication has errors. SRF10 distance sensor is working great with the OmniNXT F7 FC on same I2C bus.

## Steps to Reproduce
1. Used iNav 2.0.1 for the OmniNXT F7 FC and the iNav Configurator 2.0
2. Connect the VL53L0X SDA, SCL, 3V, GND to OmniNXT F7 FC. Add 2k2/4k7/10k pull up resistors from SDA to 3V and SCL to 3V. Use default I2C1 accordingly to the labels on the FC.
3. Plug in Lipo
4. Connect the FC to iNav Configurator 2.0 and set VL53L0X as distance sensor. Save and Reboot.
5. 5x I2C errors shown in iNav Configurator 2.0 and Sonar icon is red colored at the top.
6. Set i2c_speed =100KHZ/200KHZ/400KHZ, also not working.
7. Tried with:
[CJMCU 530 - VL53L0X](https://www.banggood.com/de/CJMCU-VL53L0X-Laser-ToF-Time-of-Flight-Ranging-Sensor-Module-For-Arduino-p-1103114.html?cur_warehouse=CN)
[Adafruit VL53L0X](https://www.adafruit.com/product/3317)
[Pololu VL53L0X](https://www.pololu.com/product/2490)
8. Now I am using SRF10 ultrasonic distance sensor on the same I2C bus with a level shifter and it is working great.

## Expected behavior
CJMCU 530 VL53L0X works with OmniNXT F7.

## Suggested solution(s)
Seems to hang in the deviceDetect method in rangefinder_vl53l0x.c. Seems the first I2C call hang, so maybe it is a bug in bus_i2c_hal.c. Tried to increase the I2C_DEFAULT_TIMEOUT to 100000, but still not works.

---

- FC Board name and vendor: [Graupner OmniNXT F7](https://www.graupner.com/Flight-Control-OmniNXT-F7-mit-OSD-BF/48377.NXT/)
- INAV version string: INAV/OMNIBUSF7NXT 2.0.1

SaschaIoT, 2019-01-13:
To get CJMCU 530 VL53L0X to work you can remove as a workaround the following in the "inav/src/main/drivers/bus_i2c_hal.c" file.

Remove in method "bool i2cWriteBuffer(I2CDevice device, uint8_t addr_, uint8_t reg_, uint8_t len_, const uint8_t *data)":

`
if (reg_ == 0xFF)
        status = HAL_I2C_Master_Transmit(&i2cHandle[device].Handle, addr_ << 1, (uint8_t *)data, len_, I2C_DEFAULT_TIMEOUT);
    else
`

and

Remove in method "bool i2cRead(I2CDevice device, uint8_t addr_, uint8_t reg_, uint8_t len, uint8_t* buf)":

` 
if (reg_ == 0xFF)
        status = HAL_I2C_Master_Receive(&i2cHandle[device].Handle,addr_ << 1,buf, len, I2C_DEFAULT_TIMEOUT);
    else
`

But attention: I wrote with @digitalentity and he sayed me, that this breaks support of some I2C sensors like for Airspeed. So do it on your own risk.

digitalentity, 2019-01-23:
Fix ready, closing

closed 2019-01-23 by digitalentity

armsp, 2022-03-08:
Hello good devs, has there been a change that can potentially make inav not to recognize VL53L0X anymore? Because I have not been able to get mine to work at all - https://github.com/iNavFlight/inav/issues/7879