iNavFlight/inav #4140: No osd after font upload

mfish33, 2019-01-02:
I am using the Matek f405ctr and was having dashed lines across my osd.  To fix this I tried a font upload but did not have the battery plugged in.  Now my osd does not show up at all even after multiple tries to upload the font with the battery and even completely re-uploading inav.

teckel12, 2019-01-02:
@mfish33 Flash INAV with a complete memory wipe as well?

mfish33, 2019-01-02:
Yeah I tried it.  Did not change anything.

teckel12, 2019-01-02:
@mfish33 Uploading the font is the correct solution for fixing the dashed lines across the screen problem. Battery disconnected should make no difference (I almost always update my FC including the font with no battery attached).

I would suggest flashing Betaflight (with full wipe) and the Betaflight font to see what happens. If it still doesn't work, maybe the OSD chip is trashed.

mfish33, 2019-01-02:
All of a sudden it started working again.  No clue why just rolling with it.  Thanks for all the help.