iNavFlight/inav #4149: Inertial Navigation System

EnsignCrasher, 2019-01-03:
Hi Guys, can someone tell me if there is some functionality in iNAV like a GPS-aided inertial navigation system? To keep track of an estimated position if GPS would somehow fail for some time?

shellixyz, 2019-01-03:
It does position estimation but in case of GPS failure the position estimate is not trusted and it will execute an emergency landing.

imrj, 2019-01-04:
INS navigation is primary navigation system on real commercial aircraft, not GPS...that being said you are asking for a lot and requires very accurate sensors (I think Boeing uses like 3 independent redundant INS systems)....but I think they are working on something over on the ardupilot software for navigation w/o GPS...
http://ardupilot.org/copter/docs/common-pozyx.html

digitalentity, 2019-01-04:
INAV is built around inertial position estimation (just like big aircrafts), however quality of sensors and conditions we're flying in are quite different, so raw estimate is only trusted for a few seconds to cover very short GPS gaps.

closed 2019-01-04 by digitalentity

EnsignCrasher, 2019-01-05:
Thanks for the quick responses! So if I got that right, when gps signal is lost, the inav will perform an emergency landing after a few seconds... Is there any way to keep it flying anyways? 
Maybe with nav cruise or anything like that?

EnsignCrasher, 2019-01-06:
Ah okay, so failsafe gets executed on GPS failure hm? Is that always the case or just for autonomous flight modes that require GPS? Wouldn't make much sense if you are just flying in angle mode and suddenly it triggers failsafe because of GPS failure...

shellixyz, 2019-01-06:
@teckel12 It doesn't depend on the failsafe settings.

@EnsignCrasher It will only execute an emergency landing if you lose fix when you're in a mode that relies on the GPS position (RTL, WP). By default after 5s (`nav_position_timeout` setting).