iNavFlight/inav #4151: Inav 2.0.1 when Bluetooth is connected to EZ GUI and when transmitter is on no signal is received

RichardAnkur, 2019-01-04:
Hello 
I have the omnibus f4 pro v2 flight controller board (black one) which I had ordered from https://robu.in/product/omnibus-f4-v2-pro-flight-controller/
I have connected my ublox neo m8n GPS and compass on UART 6 and UART 3 and I am using a flysky fs i6x transmitter and fs iA6B receiver which is connected to sbus/ppm pin using ppm.The firmware which I had uploaded is the omnibus f4pro...and not that ledstrip one.
And now I have connected my HC06 Bluetooth module to UART 1.
The problem is that when I plug in the battery and connect the bluetooth with EZ GUI app it works perfectly but as soon as I turn on my transmitter.....the female voice from EZ GUI tells data not received.
Suggest me a solution

RichardAnkur, 2019-01-05:
At UART 1 i had connected my Bluetooth module NAD msp is activated on that port with a baud rate of 11260

wx4cb, 2019-01-05:
i would suggest trying a different uart it sounds like uart 1 is also the one used for PPM maybe? I use sbus & crsf so not sure which port the ppm input is on

RichardAnkur, 2019-01-05:
My UART 6 and UART 3 is reserved for GPS. Only UART 1 is free

wx4cb, 2019-01-05:
have you tried soft serial?

RichardAnkur, 2019-01-05:
Means you are telling me to use the two pins RX and tx above moto_vcc .......
Connect RX from Bluetooth to tx in fc and tx from Bluetooth to RX in fc......and then I will get a additional option below uart6 named softserial....

shellixyz, 2019-01-06:
If you want to use UART1 on the Omnibus F4 PRO v2 for something else than S.Bus you need to use PPM for the receiver and remove the S.Bus jumper.

[image]

DzikuVx, 2019-01-08:
The solution by @shellixyz is a correct one, it is even mentioned in the docs: https://github.com/iNavFlight/inav/blob/master/docs/Board%20-%20Omnibus%20F4.md

Closing, thanks guys

closed 2019-01-08 by DzikuVx

RichardAnkur, 2019-01-12:
Yes @shellixyz is correct but  now I have connected my Bluetooth to the softserial pad and receiver to UART 1 and it working perfectly.
Thank you for helping me out