iNavFlight/inav #4152: Matek F405 CTR Barometer Unavailable

dankuz1982, 2019-01-04:
Hey guys, hate to be a bother. I'm new to iNav but I've done a lot of reading to prepare myself. I'm setting up my Matek F405-CTR board and I get this. And yes I've tried multiple chip erase full reinstalls of iNav.

# version
# INAV/MATEKF405 2.0.1 Oct 31 2018 / 19:24:04 (a04e68aaa)

# status
System Uptime: 26 seconds
Current Time: 2019-01-04T04:45:57.182+00:00
Voltage: 11.10V (3S battery - OK)
CPU Clock=168MHz, GYRO=MPU6000, ACC=MPU6000
STM32 system clocks:
  SYSCLK = 168 MHz
  HCLK   = 168 MHz
  PCLK1  = 42 MHz
  PCLK2  = 84 MHz
Sensor status: GYRO=OK, ACC=OK, MAG=NONE, BARO=UNAVAILABLE, RANGEFINDER=NONE, OPFLOW=NONE, GPS=OK
SD card: Startup failed
Stack size: 6144, Stack address: 0x10010000, Heap available: 936
I2C Errors: 5, config size: 4255, max available config: 131072
ADC channel usage:
   BATTERY : configured = ADC 1, used = ADC 1
      RSSI : configured = ADC 3, used = none
   CURRENT : configured = ADC 2, used = ADC 2
  AIRSPEED : configured = none, used = none
System load: 1, cycle time: 1002, PID rate: 998, RX rate: 49, System rate: 9
Arming disabled flags: HWFAIL RX CLI

The Barometer says unavailable even though I have the BMP280 selected in the configuration tab. Some reading has indicated to me that the LED pad and the I2C share the same pads and that the LED needs to be disabled for the barometer to work. However there's no resource mapping in iNav like there is in Betaflight.

How have people gotten the Baro to work? My indicator at the top of the screen is showing red. I've attached a full dump. Keep in mind that this isn't fully set up yet either, so some things may not appear correct. I want to get all of my sensorts working first before I do any other tuning and such.

[DJIF450-MatekF405-CTR-dump Jan4-2018.txt](https://github.com/iNavFlight/inav/files/2728514/DJIF450-MatekF405-CTR-dump.Jan4-2018.txt)

giacomo892, 2019-01-04:
I see you have 5 I2C errors. 
Do you have other peripherals connected to the bus?

dankuz1982, 2019-01-04:
Oh man I'm so dumb. I had the Smartport wire there on the LED pad because that's the pad I used for Softserial when this was a BF board. Removed it and now it works. I don't know how I didn't see that, or realize it. Thanks so much! I'll close this topic.

closed 2019-01-04 by dankuz1982