iNavFlight/inav #4163: Blackbox logging on MATEK F405

Elbarto, 2019-01-07:
## Current Behavior
I Have trouble with the blackbox logging when a log is already present in the SD card. Here are the step I do to reproduce the behavior.

## Steps to Reproduce
1. Format SD card with appropriate SD card software
2. Plug SD card in the quad
3. Do a fly, shutdown everything, remove SD card and copy log.
4. Open log with the blackbox viewer from inav, everything works fine.
5. Connect the quad to the configurator to adjust filtering parameters (Thanks to the logs).
6. Disconnect the quad and plug back the SD card (without removing logs or anything).
7. Do a second fly.
8. Try to open the the new logs with blackbox viewer and I get this error : 
             Error opening log: No logs in this file could be parsed successfully

## Additional context
I've try to copy logs from a MAC and a WINDOWS to see if the operating system was messing up with the SD card without any good conclusion. 
My SD CARD is a Kingston 8GB SDHC class 10.
I guess INAV support only 4GB SD card because a file called FREESPACE.E is created when I plug it into the quad, so maybe that is causing an issue ? (I don't have 4GB SD card to do the test)
At first I thought the logging rate was to high, but even at 20% (1/5) for a 4Khz Loop, the issue is still here.

CLI dump here : https://pastebin.com/w96Wd0sy

Working log of the 1st flight : 
[LOG00001.TXT](https://github.com/iNavFlight/inav/files/2733316/LOG00001.TXT)

Bugged log of the second flight : 
[LOG00002.TXT](https://github.com/iNavFlight/inav/files/2733324/LOG00002.TXT)

I did that test 5 time, and every time the seconds logs (and everyone that came after) are corrupted.

---

- FC Board name and vendor: MATEK F405-CTR
- INAV version string: INAV/MATEKF405 2.0.1 Oct 31 2018 / 19:24:04 (a04e68aaa)

stronnag, 2019-01-07:
I have no problem with multiple logs to suitable SD cards on MATEKF405.
I also have what should be a a suitable card that refused to write any sane log (just megabytes of  NUL bytes) on the same platforms. That card is now related to more mundane tasks like camera video.

Elbarto, 2019-01-09:
I've try today with a Sandisk ultra 16GB and it worked fine without problem. I guess it's better to stick to the recommended SD card.

closed 2019-01-09 by Elbarto