iNavFlight/inav #4165: Quad won't start mission

Cecemel20, 2019-01-07:
Hi,
I planned a mission using iNav Configurator, saved to EEPROM (I tested by trying to 'load from EEPROM', so I know it's saved). Turning on WP mode doesn't do anything. I had 17 sats, PosHold, AltHold and RTH work perfectly and I was close to the first waypoint. I'll include a dump and blackbox log.

FC: Matek F405-CTR
iNav Version: 2.0.1 (Dump says 2.1.0 as @teckel12 made some I2C address changes to get my mag working #4131 )
[LOG00017.TXT](https://github.com/iNavFlight/inav/files/2733715/LOG00017.TXT)
[dump.txt](https://github.com/iNavFlight/inav/files/2733727/dump.txt)

stronnag, 2019-01-07:
No waypoints in volatile memory .....
```
H waypoints:0,0
```

stronnag, 2019-01-07:
Behaviour entirely predictable

Cecemel20, 2019-01-07:
@stronnag Then what did I do wrong in mission planning? I put down the waypoints, pressed save to FC, pressed save to EEPROM.

When I connect the quad to my laptop, go to mission control, press Load mission from FC or Load Eeprom mission (both work), the waypoints show up on the map. So they must be stored on the FC?
What’s the difference between Eeprom and FC actually?

Cecemel20, 2019-01-07:
Ok, thank you.

closed 2019-01-07 by Cecemel20

Heli-Ebook, 2019-12-12:
I've had  the SAME issue and been hacking away at it for 2 days now.
So.... LONG STORY SHORT..... ALL the videos on YouRube are an OUTRIGHT LIE! I figured as much, one of them was posted by Paweł Spychalski, man... what a BULLSHIT ARTIST!  You've lost any respect you (never had anyway) from me buddy, I've never even seen you FLY anything in your videos! List of Youtube bullshit artists below (I'm not at all surprised either, people will kill their grandma if there's enough cash in it!)

"Blue Falcon Project" (some child in mommies basement)
iForce2D (who IMO accepts too many "gifts" from BangGood to be trusted)
"JD"
"Painless 360" (another Bozo who leaves out critical; steps, like mixer presets.
Paweł Spychalski

WHAT A BUNCHA W H O R E S !

Heli-Ebook, 2019-12-12:
OH YEAH, another thing that SCREWED ME last year: Nobody bothered to mention the need for loading the CP210x USB drivers ANYWHERE, not in any YouRube video nor on an OPENING SCREEN of iNav! The iNav docs don't mention anything about the need for "volatile memory" anywhere on the Waypoints page, so you've FAILED at a level a FIFTH GRADER fails!

Ciao

Heli-Ebook, 2019-12-12:
OH.... do I have to enable stick programming in iNav now also?

"Oops.... we forgot THAT too"!  You guys probably can't satisfy your wives either!

Heli-Ebook, 2019-12-12:
No, you don't need stick programming and 
set nav_wp_safe_distance = 65000 in CLI
whittle it down from there BUT 65k allows your arming switch to function
after reading in the WP's. Flipping NAV WP afterwards makes motors respond.

As usual, gotta figure it out for yourselves! You're welcome.