iNavFlight/inav #4169: TFMini Support?

CradyLab, 2019-01-07:
I see that the TFMini is absent in 2.1.0 -- anyone know when/if it will be supported?

DzikuVx, 2019-01-07:
You mean `2.1.0` or `2.0.1`? Because TFmini will be supported in `2.1.0`

CradyLab, 2019-01-07:
Thanks, Pawel.  I just downloaded Configurator 2.1.0 and thought that was it

DzikuVx, 2019-01-07:
We did not relased the firmware for INAV 2.1, you would have to wait a day or so. Also it will be supported only on F4 and F7 boards

closed 2019-01-07 by DzikuVx

CradyLab, 2019-01-07:
Awesome.  I have Matek 722 Std ready for it =)

Thanks so much for the reply, Pawel.  Much appreciated!

shellixyz, 2019-01-07:
You won't be able to configure the rangefinder from the GUI with this release. Should be fixed in 2.1.1. For now you will need to do it from the CLI. The function is 65536.

Example: `serial 2 65536 115200 115200 0 115200`

CradyLab, 2019-01-07:
Merci, Michel!  Thanks for the heads-up