iNavFlight/inav #4172: About the uart data receiving for RunCam Split

azolyoung, 2019-01-08:
## Current Behavior
I am changing the code for the runcam device communication to asynchronous, which has been implemented and tested on Betaflight and works fine. But when I moved it to iNav, I found that when I received data from split, it was always wrong or incomplete. This wrong situation is very similar to the incorrect BaudRate, but I confirmed that the Baud parameter of openSerialPort is 115200.

I suspect that the options may be incorrect, So I added SERIAL_STOPBITS_1 | SERIAL_PARITY_NO and the result is the same. 
Later, I tried to delay the call of openSerialPort by 3 seconds, 10 seconds (because I was worried that the runcam device might take a little time to initialize) and send the command, the situation remains the same, the received data is incomplete or incorrect.

However, it is ok to send commands. Sending control commands to the camera  and then the camera  will response normally , such as simulate the  Power Button, Wi-Fi Button, and the camera can respond correctly. I also tried to communicate with RunCam Split using the USB-TTL Adapter, which gives the correct response when I send "cc 00 60" to it. "cc 00 60" is a query command, normally, it will return 5 bytes data the including the protocol version and features.

To facilitate debugging, I used DEBUG_TRACE and mounted it to one of my free serial port to observe the data (it has to be said that this function is really easy to use, much easier than the debugInfo array).

The same set of test equipment (flying control, RunCam Split, etc), the code is almost the same (except that the serialWriteBuf is not the same as the call of BF), they can work normally in BetaFlight, and now I suspect that there is something wrong for the iNav serial port reading function. What should I pay attention to?

## Steps to Reproduce

1. Power on the quad
2. Waiting about 10 seconds(it's a test code)
3. get the output of the DebugTrace on the serial tools, you will see the response of runcam split after 0xcc 0x00 0x60 is incorrect, sometime the response is 0x00 0x00, sometime it's 0x00 0x00 0x04 0x00 0x00, in this case, the length of response is correct, but the content not.

## Expected behavior
after sent "cc 00 60" to RunCam Split, and read from split(serialRead or rxCallback in openSerialPort), it should get 5 bytes response, and the content is "CC 01 57 00 B7"

## Suggested solution(s)
No idea for now

## Additional context
The following two images, one is the output of the DebugTrace, as you can see, the response data after "0xcc 0x00 0x60", the response is incorrect.
https://s3.amazonaws.com/rctmpfiles/Screen+Shot+2019-01-08+at+16.42.36.png
and the other image is showing you the correct one, when I send CC 00 60, the response is "CC 01 57 00 B7", that is correct one. the data was sent through USB-TTL adapter to RunCam Split, and runcam split was power by USB.
https://s3.amazonaws.com/rctmpfiles/Screen+Shot+2019-01-08+at+17.16.44.png

- FC Board name and vendor:
SpeedyBee F405

- INAV version string: 
The newest code from developer branch.

fiam, 2019-01-16:
@azolyoung can you join our telegram group at https://t.me/INAVFlight ?

azolyoung, 2019-01-17:
Yes,  I have joined the grou, wow you guys are playing on there!

closed 2019-04-01 by stale[bot]