iNavFlight/inav #4173: 2.1.0 RC1 - VTX OSD power level indicator not updated on first arm

putimir, 2019-01-09:
I'm testing the 2.1.0 RC1 on one of my wings. I have set `feature FW_LAUNCH` and configured VTX (AKK X2) for SA control, with 

```
set vtx_power = 4
set vtx_low_power_disarm = ON
set vtx_freq = 5805
```
...meaning in the disamed state the VTX is in power level 1, and on arm it switches to power level 4.

Upon first arm, entering autolaunch, the power level is correctly bumped to power level 4, as it is displayed on the vtx's display, but the OSD element remains 1. 
If I then disarm, and arm again, the OSD element switches correctly to 4.

I'm using this feature for some time now, and never had such behaviour. This model was ok regarding this issue, up to the update last night.

I tried different baud rates on the UART, but no change.

Anybody else having this? It's not that big of a deal, I'm having this in the osd for reassuring the vtx is in full power before I go off, but I reported this, as I think it is quickly fixable.

Arvycka, 2019-01-19:
I was about to pen a new issue "2.1.0-RC1 VTX OSD not updated in real time over SmartAudio" but this was suggested by the system.
I will add that VTX OSD is not being updated (power and channels do change correctly) at all without rebooting the FC. After the reboot the changes are displayed correctly in OSD - with "low power while disarmed" disabled.
With the above feature enabled - power is bumped to the preset level on arming, OSD remains on low setting. On disarm OSD switches to display high level.

I have tested the same receiver with INAV 2.0.1 and everything is working correctly there - VTX setting changes are updated immediately on screen.
https://youtu.be/3DIc7mHYoB0

giacomo892, 2019-01-20:
What flight controllers are you running so I can build you a test firmware?

Arvycka, 2019-01-20:
I can test on Omnibus F3

Arvycka, 2019-01-20:
Just tested - it's working correctly now, thanks. I assume this wasn't specific hardware related?

giacomo892, 2019-01-20:
I think it is not. We fixed the issue yesterday and it will be part of RC2. So you can close the issue now.

putimir, 2019-01-20:
Hi, I'm running matekf405, but if you're sure you got it, thank you very
much! Is the fix merged with the development branch?

giacomo892, 2019-01-20:
Yes it is.

closed 2019-01-20 by giacomo892