iNavFlight/inav #4178: 2.1.0 RC1 delayed GPS acknowledgement time

kardon18, 2019-01-09:
**Target** -    _**Omnibus F4 Pro**_
**GPS module -**     _**Matek SAM M8Q + 5883**_
**Firmware version** -    _**Inav 2.1.0 RC1**_

This was confirmed by the configurator and a stop watch.
Both these tests were confirmed with both firmware's uses both and alternate 2.0.0 and 2.1.0 configurators. _Not just with the configurator designed for its firmware._ I wanted to be sure were the problem was.

**Inav 2.0.1 operation -**
When the flight controller is initialised and power.  
The GPS will take 3 seconds to get acknowledgement from the flight controller. 
Then 1.5 seconds to hookup and start transferring data.
It received 4 timeouts.

[image]

**Inav 2.1.0 operation -**
When the flight controller is initialised and power.  
The GPS will take 10 seconds to get acknowledgement from the flight controller. 
Then 43.5 seconds to hookup and start transferring data.
It received 24 timeouts.

[image]

I ran these tests 3 times. Back and fort between flashes. And 6 times times total, between both configurators.
And the inav to GPS acknowledgement time was precisely 10sec, and precisely 43.5 sec before inav would acknowledge the data.
That sounds to precise to me, to just be chance.

leeph, 2019-01-11:
I flashed 2.1.0 RC1 to two craft yesterday, one multirotor and one flying wing running Matek F722-STD.

Multirotor all fine, but on the flying wing the GPS goes red shortly after boot-up and the beeper sounds at 2hz. I may not have left it for 45 seconds to see if it would resolve itself, but I'm wondering if this is the same issue. 

Does your GPS go red at all after power-on? If not, I may just have disturbed the wiring and need to re-solder.

Multicopter has Beitian BN-880 while the flying wing has the Matek GPS+MAG unit. Both running UBLOX7 protocol.

kardon18, 2019-01-11:
@leeph 
In the first case I displayed with 2.0.1. It goes RED on the configurator after 3 seconds. Then turns BLUE after another 1.5seconds. Then it starts sending message between the GPS and FC.

But with 2.1.0. RC1;  It takes 10 seconds to go RED. Then another 43.5 seconds before it turns BLUE, and start sending message between the GPS and FC.

Its way slower than before.  
This is not including the time that follows after this; before it gets satellite acquisition.

leeph, 2019-01-11:
Confirmed, GPS goes red on startup and the beeper goes mad, then after about half a minute or so it turns blue and everything seems stable.

I feel this is something to do with the GPS auto config and change to 115200bps, but logging seems practically non - existent in inav so difficult to say anything for sure.

DzikuVx, 2019-01-15:
fixed in #4193

closed 2019-01-15 by DzikuVx

leeph, 2019-01-15:
Thanks for your efforts, guys.

tn2009, 2019-02-26:
Flashed 2.1.0 and Now the GPS light is steady red (not blue lights) and the drone beeping. There was a firmware 1.9.1 on it everything was fine. How to fix?

leeph, 2019-02-26:
I've flashed 2.1.0 stable and the M8Q GPS has been detected and configured in a timely fashion on all so far. Odd behaviour, up to this point I was having similar issues but no more it seems.

tn2009, 2019-02-26:
2.1.0 RC1 RC2 RC3 and 2.1.0 stable have an error. GPS light is steady red.

wx4cb, 2019-02-26:
i've got 2.1 on 3 aircraft (3 different FCs) and they all are working as they should with the gps. as @leeph i had an issue on at least one of these and they (so far) appear to not have this issue in my limited testing since release. 

I will test later this week when i can get out and actually fly them. but right now i get a good lock from a complete cold start in about 1.5 minutes whereas before at least one of them was taking 5 minutes

Nihilistic, 2019-02-26:
Flashed 2.1 full on my 5" quad with the Matek F722-SE and BN-880 GPS. The GPS indicator on the configurator is off at first when plugging in USB and immediately connecting to iNAV (after being fully configured), then after about 5 seconds it goes red, and then another 10 seconds or so it goes blue. And then it takes a few minutes to get a 3D lock with HDOP under 2 on the bench inside the house (but that's normal).

stronnag, 2019-02-26:
> Flashed 2.1.0 and Now the GPS light is steady red (not blue lights) and the drone beeping. There was a firmware 1.9.1 on it everything was fine. How to fix?

Most lileky you need to check / fix your cabling; prior to 2.1 iNav did not check for acknowledgements from the GPS, now it does. This is a feature, not a bug.

PR #4193 (2019-01-12): [GPS] Change autobaud logic for UBX protocol to send baud change command in bulk
Change autobaud behavior on UBLOX/UBLOX7 protocols. Intended to fix #4178