iNavFlight/inav #4179: 2.1.0 RC1 Build error 336

wrenfpv, 2019-01-09:
## Current Behavior
I try to build following the steps in "Building in windows".

## Steps to Reproduce
1. Installed Cygwin, GNU ARM Toolchain, and Setup Ruby.
2. Opened the Cygwin-Terminal
3. cd /cygdrive/c/dev
    git clone https://github.com/iNavFlight/inav.git
4. cd inav
    make TARGET=MATEKF405SE

## Expected behavior
Within a few moments I should have the binary ready.

## Suggested solution(s)
I think the windows build documentation is not updated, anyone can help?

## Additional context
c:/dev/gcc-arm-none-eabi-8-2018-q4-major-win32/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: C:\cygwin64\tmp\ccOhMlxOdebugobjtem: invalid string offset 143654912 >= 22975072851462314 for section `(null)'
C:\cygwin64\tmp\ccp3TJMSdebugobjtem: file not recognized: file format not recognized
collect2.exe: error: ld returned 1 exit status
lto-wrapper.exe: fatal error: C:\Dev\gcc-arm-none-eabi-8-2018-q4-major-win32\bin\arm-none-eabi-gcc.exe returned 1 exit status
compilation terminated.
c:/dev/gcc-arm-none-eabi-8-2018-q4-major-win32/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: error: lto-wrapper failed
collect2.exe: error: ld returned 1 exit status
make: *** [Makefile:336: obj/main/inav_MATEKF405SE.elf] Error 1

---

N/A
- FC Board name and vendor: N/A
- INAV version string: N/A

datamann, 2019-01-09:
@wrenfpv
Another alternative could be to use docker
https://github.com/iNavFlight/inav/issues/3896#issuecomment-426416539

tiriad, 2019-01-13:
Also create a virtual machine and install linux to build the version from there with the manual for that platform

digitalentity, 2019-01-13:
Yes, there seem to be a bug in gcc 8.2.1 under windows. Earlier versions compile fine

closed 2019-02-02 by shellixyz