iNavFlight/inav #4183: Matek F405-CTR OSD blincking on 2.1.0

limayankee, 2019-01-09:
## Current Behavior
Update to 2.1.0 on Matek F405-CTR and the OSD blinks, like it's a refresh problem.
Had no problems with 2.0.1.
Video: [https://youtu.be/F_L0ycSTDic](url)

---
- FC Board name and vendor:  Matek F405-CTR
- INAV version string: INAV/MATEKF405 2.1.0 Jan  8 2019 / 13:32:43 (f8de2e6a3)

Nihilistic, 2019-01-13:
I have the F405-CTR, and I'm running 2.1 RC1. Not having this blinking OSD issue at all. 

Did you check "full chip erase" when you flashed it? Did you paste dumps or use diffs to configure it when you set it up, or do it all from scratch? Did you load the font first on the OSD tab? Do you have any OSD alarms set? Try setting all those to zero and not using them as a workaround maybe?

limayankee, 2019-01-13:
I tired every things you said. 
I did the full erase with from scratch, from a dump and from a diff.
I tired to update the font as well with 3 different fonts.
Yes I have alarms set. I will try to disable the alarms tomorrow and I will let you know.

shellixyz, 2019-01-13:
I doubt this is related to alarms. Strange. Looks like a refresh triggered at 1Hz

Nihilistic, 2019-01-13:
@ limayankee, is your FPV camera NTSC or PAL? Select whichever one you're using and click save on the OSD tab, rather than leaving it on auto.

limayankee, 2019-01-14:
@shellixyz I had the same thought

limayankee, 2019-01-14:
@Nihilistic I tried everything you said and nothing change. The same issue. 
Flash back to 2.0.1 and there it's works perfectly.

Nihilistic, 2019-01-14:
What's your VTX and camera? Getting power for both from the board? If so, from which pins? Or directly from the flight battery? Running 4S?

Nihilistic, 2019-01-14:
Are any of the fields on your OSD overlapping? Any fields active but not showing on the OSD preview window? I think it may be because you have some fields up inside the side bars and beneath the artificial horizon. Try moving everything outside or above or below the side bars, and no fields running off the sides of the screen that don't fully fit.

limayankee, 2019-01-14:
@Nihilistic everything works with 2.0.1. It's a software issue.
For the info the cam and vtx are powered directly from the battery.
The cam is an eagle 2 pro and the vtx FFPV 2.5.
Running with 3S.
I use the TBS Grandstation.
I think it's a refresh issue, maybe between the osd and the screen refresh rate.

limayankee, 2019-01-14:
@Nihilistic I manage to fix it, by changing the PAL and NTSC several times and rebooting the board several times.
No idea what causes the issue

Nihilistic, 2019-01-14:
Did you leave your OSD layout the same? You can't have fields that overlap, like stuff where the artificial horizon travels. Don't have anything inside the sidebars. Or sometimes even having too many fields on the same line will cause weird issues. Like, the way you have yours, when you pitch back, your artificial horizon is going to conflict with your altitude and speed field, and one or the other gets wiped out.

limayankee, 2019-01-14:
@Nihilistic Yes I restore from a dump so the same as the initial video.
It's was something with the PAL/NTSC

Nihilistic, 2019-01-14:
Ok, well try tilting your copter back so the artificial horizon conflicts with those fields you have and see what happens. You're going to need to move that stuff or turn artificial horizon off.

DzikuVx, 2019-01-15:
This is NTSC detection bug will be addressed before RC2

fiam, 2019-01-15:
@limayankee Can you test with the hex I've posted in https://github.com/iNavFlight/inav/pull/4208?

limayankee, 2019-01-16:
@fiam I do the full cheap erase flash and it's works with the auto selected in the osd tab.

shellixyz, 2019-02-02:
Fixed, closing

commit cc6baedf (2019-01-25): Fix detection of NTSC signal after 1.5s

Due to a logic error during refactor, signal was always being stored
internally as PAL when more than 1.5s had elapsed since boot. The
MAX7456 kept reporting that the signal was actually NTSC, causing the
reinitialization sequence to trigger and the OSD to blink once every
second, but since the data cached in RAM was always stored as PAL
this continued forever.

Fixes #4183
closed 2019-02-02 by shellixyz