iNavFlight/inav #4189: iNAV 2.1 RC1 on Matek F722-SE apparent dsync using DSHOT 600

Nihilistic, 2019-01-10:
## Current Behavior

Maiden flight of 5" quad abruptly concluded upside down in the grass 30 seconds into the flight. Everything seemed fine, and suddenly the motors just shut off. Luckily I was only 20 feet up and landed in grass. 
## Steps to Reproduce

1.Flashed iNAV 2.1 RC1 to the FC with full chip erase selected. 
2.Selected 5" Performance preset, and DSHOT 600
3.Finished the rest of the configuration manually (no diffs or CLI pastes)
4.Flew with all default PIDS & Rates

## Expected behavior

A normal pleasant flight without falling out of the sky lol
## Suggested solution(s)

Something may be wrong with the DSHOT code in iNAV. I flew the same ESC's using DSHOT600 on betaflight 3.5.X (but this flight was the maiden for the F722-SE).
## Additional context

F722-SE flight controller
Individual Spedix BLheli_S 25A ESC's with 16.7 firmware.
Gemfan 2205 2300kv motors
King Kong 5040 biblade props
Props and motors are in like new condition, not bent, bearings are fine. Never been crashed until this incident.
https://pastebin.com/CQT6mhdx
[LOG00022.TXT](https://github.com/iNavFlight/inav/files/2747293/LOG00022.TXT)

---

- FC Board name and vendor:
Matek F722-SE purchased from Helidirect.com

- INAV version string:
# INAV/MATEKF722SE 2.1.0 Jan  9 2019 / 09:13:14 (f8de2e6a3)
# GCC-7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]

teckel12, 2019-01-11:
#4181 Possibly related

digitalentity, 2019-01-11:
DSHOT code is fine and well tested, many flight-hours with DSHOT by @DzikuVx et al.

Bear in mind that DSHOT is not a high-reliability protocol, in fact it's much less reliable than old-school analog protocols. The fact that it is flyable in Betaflight doesn't justify for it's low reliability in general. Betaflight "fixes" the issue by much more retransmissions hoping that some packets will get through but that doesn't mean that it's more reliable there. 

Mitigation is to not use DSHOT600 and DSHOT1200. DSHOT150-300 are reasonably reliable.

closed 2019-01-11 by digitalentity

Nihilistic, 2019-01-11:
@digitalentity,

Would you say DSHOT enhances flight performance enough over analog protocols to be worth the trade off in reliability? I'm willing to give DSHOT 150 and 300 a try if it is. But, I'll take the hassle of removing props to calibrate ESC's and putting them back on, over slogging through a soggy field for a mile or two to recover my quad any day, if not ;)

digitalentity, 2019-01-11:
Regarding flight performance/reliability tradeoffs - everybody decides for himself. I'd say DSHOT150 has similar reliability to Multishot. Frankly, I believe for GPS-enabled machines difference between protocols is quite negligible.

lunebleue-zz, 2019-03-03:
Had the same issue today with the final realease 2.1.0, with Matek F722-SE as well and **DSHOT 300**, all 4 motors suddenly just shut off. The flights done with Betaflight with DSHOT 300 no such problem !!!  Hardware + settings seems not in cause in my case... Will go back to analog signal with INav until this issue is solved...

shellixyz, 2019-03-03:
@lunebleue-zz Have you got a BB log ?

lunebleue-zz, 2019-03-03:
Unfortunately not when it happens in flight, after that as there was no important damages (except GPS mount broken) I have juste made a few arming and disarming on the ground (with GPS hanging on the side) and during that I had once the 4 motors stopping as well (in my OSD all was like it was not stopped (no resume screen like when you Disarm) In the log bellow (made on the ground afer the crash there could be the issue recorded or not I don't know as I had black box ON not all the time...  If it can help here they are...
[LOG00012.TXT](https://github.com/iNavFlight/inav/files/2923626/LOG00012.TXT)
[LOG00009.TXT](https://github.com/iNavFlight/inav/files/2923627/LOG00009.TXT)
[LOG00010.TXT](https://github.com/iNavFlight/inav/files/2923628/LOG00010.TXT)

lunebleue-zz, 2019-03-03:
All 4 motors at once stopped doesn't sound like noise on my quad perturbing digital signal and furthermore with DSOT 300 and just spinning on the ground with short and twisted wire for the ESC signal and good soldering.
ESC are brand NEW DYS ARIA 35A 3-6s BLHELI_32 (4s setup for me).

digitalentity, 2019-03-03:
This is a closed issue. Please open a new one and reference this one